liberatedsystems/RNode_Firmware_CE #5: Custom rnodeconf build

jacobeva, 2024-05-20:
I need to create an rnodeconf-like utility which will support all the extra boards which are able to be flashed by this repository.

WifiHero91, 2024-10-12:
import os
import re
import subprocess

MAKEFILE_PATH = "Makefile"

def parse_firmware_targets(makefile_path):
    """Parse firmware targets from the Makefile."""
    with open(makefile_path, "r") as file:
        content = file.read()

    # Regex to find firmware targets
    firmware_pattern = re.compile(r"^firmware-([\w_]+):", re.MULTILINE)
    targets = firmware_pattern.findall(content)
    
    return targets

def parse_upload_targets(makefile_path):
    """Parse upload targets from the Makefile."""
    with open(makefile_path, "r") as file:
        content = file.read()

    # Regex to find upload targets
    upload_pattern = re.compile(r"^upload-([\w_]+):", re.MULTILINE)
    targets = upload_pattern.findall(content)
    
    return targets

def select_target(targets, target_type):
    """Allow the user to select a target from the list."""
    print(f"\nAvailable {target_type} targets:")
    for idx, target in enumerate(targets):
        print(f"{idx + 1}: {target}")

    choice = int(input(f"\nSelect a {target_type} target by number: ")) - 1
    return targets[choice]

def build_firmware(firmware_target):
    """Build the firmware using the selected firmware target."""
    try:
        print(f"Building firmware for: {firmware_target}...")
        subprocess.check_call(["make", firmware_target])
        print(f"Firmware {firmware_target} built successfully.")
    except subprocess.CalledProcessError:
        print(f"Error occurred during build of {firmware_target}.")
        exit(1)

def upload_firmware(upload_target):
    """Upload the firmware using the selected upload target."""
    try:
        print(f"Uploading firmware to: {upload_target}...")
        subprocess.check_call(["make", f"upload-{upload_target}"])
        print(f"Firmware uploaded successfully to {upload_target}.")
    except subprocess.CalledProcessError:
        print(f"Error occurred during upload to {upload_target}.")
        exit(1)

if __name__ == "__main__":
    # Step 1: Parse firmware and upload targets
    firmware_targets = parse_firmware_targets(MAKEFILE_PATH)
    upload_targets = parse_upload_targets(MAKEFILE_PATH)

    if not firmware_targets:
        print("No firmware targets found in Makefile.")
        exit(1)

    if not upload_targets:
        print("No upload targets found in Makefile.")
        exit(1)

    # Step 2: Select firmware and upload targets
    selected_firmware = select_target(firmware_targets, "firmware")
    selected_upload = select_target(upload_targets, "upload")

    # Step 3: Build the firmware
    build_firmware(f"firmware-{selected_firmware}")

    # Step 4: Upload the firmware
    upload_firmware(selected_upload)

jacobeva, 2024-10-12:
@wifihero91 sent me this earlier:
```
import os
import re
import subprocess

MAKEFILE_PATH = "Makefile"

def parse_firmware_targets(makefile_path):
    """Parse firmware targets from the Makefile."""
    with open(makefile_path, "r") as file:
        content = file.read()

    # Regex to find firmware targets
    firmware_pattern = re.compile(r"^firmware-([\w_]+):", re.MULTILINE)
    targets = firmware_pattern.findall(content)
    
    return targets

def parse_upload_targets(makefile_path):
    """Parse upload targets from the Makefile."""
    with open(makefile_path, "r") as file:
        content = file.read()

    # Regex to find upload targets
    upload_pattern = re.compile(r"^upload-([\w_]+):", re.MULTILINE)
    targets = upload_pattern.findall(content)
    
    return targets

def select_target(targets, target_type):
    """Allow the user to select a target from the list."""
    print(f"\nAvailable {target_type} targets:")
    for idx, target in enumerate(targets):
        print(f"{idx + 1}: {target}")

    choice = int(input(f"\nSelect a {target_type} target by number: ")) - 1
    return targets[choice]

def build_firmware(firmware_target):
    """Build the firmware using the selected firmware target."""
    try:
        print(f"Building firmware for: {firmware_target}...")
        subprocess.check_call(["make", firmware_target])
        print(f"Firmware {firmware_target} built successfully.")
    except subprocess.CalledProcessError:
        print(f"Error occurred during build of {firmware_target}.")
        exit(1)

def upload_firmware(upload_target):
    """Upload the firmware using the selected upload target."""
    try:
        print(f"Uploading firmware to: {upload_target}...")
        subprocess.check_call(["make", f"upload-{upload_target}"])
        print(f"Firmware uploaded successfully to {upload_target}.")
    except subprocess.CalledProcessError:
        print(f"Error occurred during upload to {upload_target}.")
        exit(1)

if __name__ == "__main__":
    # Step 1: Parse firmware and upload targets
    firmware_targets = parse_firmware_targets(MAKEFILE_PATH)
    upload_targets = parse_upload_targets(MAKEFILE_PATH)

    if not firmware_targets:
        print("No firmware targets found in Makefile.")
        exit(1)

    if not upload_targets:
        print("No upload targets found in Makefile.")
        exit(1)

    # Step 2: Select firmware and upload targets
    selected_firmware = select_target(firmware_targets, "firmware")
    selected_upload = select_target(upload_targets, "upload")

    # Step 3: Build the firmware
    build_firmware(f"firmware-{selected_firmware}")

    # Step 4: Upload the firmware
    upload_firmware(selected_upload)
```

Although it isn't what I'm literally looking to do, it may be useful for helping people who don't have skills in using make to flash their boards directly from the repo, instead of downloading the firmware as a binary blob from online.

cobraPA, 2025-05-10:
I'm probably going to see if going the route of web flasher is better for general use than rnodeconf.  I will likely do that to support flashing of iOS BLE variants from my repo.  (And have additional board support as well...)

jacobeva, 2025-07-25:
This is currently in the works with `rnodeman`, but it is not finished yet. Having an issue here doesn't really make sense, so closing.

closed 2025-07-25 by jacobeva