liberatedsystems/RNode_Firmware_CE #7: RAK4631 firmware problem

shodan8192, 2024-05-27:
Apart from problems mentioned in https://github.com/liberatedsystems/RNode_Firmware_CE/issues/6#issuecomment-2133097236 I've managed to compile firmware for RAK4631 and upload it to device. Then bootstrapped EEPROM by :
`rnodeconf /dev/ttyACM1 -r --platform 70 --product 10 --model 12 --hwrev 01`

`rnodeconf /dev/ttyACM0 -i` returns :
```
[11:54:35] Opening serial port /dev/ttyACM0...
[11:54:38] Device connected
[11:54:38] Current firmware version: 1.72
[11:54:38] Reading EEPROM...
[11:54:39] EEPROM checksum correct
[11:54:39] Device signature validated
[11:54:39] 
[11:54:39] Device info:
[11:54:39] 	Product            : RAK4631 779 - 928 MHz (10:12:51)
[11:54:39] 	Device signature   : Validated - Local signature
[11:54:39] 	Firmware version   : 1.72
[11:54:39] 	Hardware revision  : 1
[11:54:39] 	Serial number      : 00:00:00:07
[11:54:39] 	Modem chip         : SX1262
[11:54:39] 	Frequency range    : 779.0 MHz - 928.0 MHz
[11:54:39] 	Max TX power       : 22 dBm
[11:54:39] 	Manufactured       : 2024-05-24 21:46:17
[11:54:39] 	Device mode        : Normal (host-controlled)
```
but by `rnstatus -vvvvvv` I got :
```
...
[2024-05-27 11:23:46] [Notice] Opening serial port /dev/ttyACM0...
[2024-05-27 11:23:48] [Notice] Serial port /dev/ttyACM0 is now open
[2024-05-27 11:23:48] [Verbose] Configuring RNode interface...
[2024-05-27 11:23:48] [Verbose] Waiting for radio configuration validation for RNodeInterface[RNode LoRa Interface]...
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting frequency is 867.2 MHz
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting bandwidth is 125.0 KHz
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting TX power is 7 dBm
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting spreading factor is 8
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting coding rate is 5
[2024-05-27 11:23:48] [Verbose] RNodeInterface[RNode LoRa Interface] On-air bitrate is now 3.12 kbps
[2024-05-27 11:23:48] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting state is offline
[2024-05-27 11:23:48] [Error] Radio state mismatch
[2024-05-27 11:23:48] [Error] After configuring RNodeInterface[RNode LoRa Interface], the reported radio parameters did not match your configuration.
[2024-05-27 11:23:48] [Error] Make sure that your hardware actually supports the parameters specified in the configuration
[2024-05-27 11:23:48] [Error] Aborting RNode startup
...
```

Am I miss a step or RAK4631 is not yet usable ?

shodan8192, 2024-05-28:
Ok, I'll answer to myself. The missing step was to set firmware hash by :
`rnodeconf /dev/ttyACM0 -H FIRMWARE_HASH`
where FIRMWARE_HASH can be obtained from :
`rnodeconf /dev/ttyACM0 -L`

closed 2024-05-28 by shodan8192

jacobeva, 2024-05-29:
I apologise, I will eventually get around to writing a guide for building and flashing the firmware yourself!

I should also mention adding more `v`s to `rnodeconf -v` doesn't do anything ;)

shodan8192, 2024-05-29:
You mean `rnstatus -v` ? As per :
https://github.com/markqvist/Reticulum/blob/master/RNS/Utilities/rnstatus.py#L295
https://github.com/markqvist/Reticulum/blob/master/RNS/Utilities/rnstatus.py#L309
more `v` increases verbosity :)

jacobeva, 2024-05-29:
Ah sorry, I didn't check the tool you were using! My fault.

jacobeva, 2024-06-03:
Addressed in commit 5e234a2. Flashing automatically now works by running `make firmware-rak4631 && make upload-rak4631`.