liberatedsystems/RNode_Firmware_CE #9: Build Environment - Ed25519

starlight-traveler, 2024-06-06:
Hello!

Given that this is a project utilizes Arduino, I am unsure as to what header this library is asking for, likely external but no docs on it.

```
 #include <Ed25519.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1

```

A cursory search doesn't bring up anything too obvious, and I do not see any documentation on how to set up a build environment (though I apologize if I am wrong, just point me to it). Just wanted to test out adding a board to the current configuration and see if something will need to be done to enable the PA:

```
  #elif BOARD_MODEL == BOARD_T3_S3_V1_2_SX1280_PA
    #define HAS_BLUETOOTH true
    #define HAS_PMU true
    #define HAS_EEPROM true
    #define EEPROM_SIZE 296 // minimum EEPROM size
    #define EEPROM_OFFSET EEPROM_SIZE-EEPROM_RESERVED
    #define MODEM SX1280
    #define HAS_BUSY true
    #define HAS_RF_SWITCH_RX_TX true
    const int pin_cs = 7;
    const int pin_reset = 8;
    const int pin_rxen = 21;
    const int pin_txen = 10;
    const int pin_dio = 9;
    const int pin_busy = 36;
    const int pin_tcxo_enable = -1;
    const int pin_led_rx = -1;
    const int pin_led_tx = 37;
```

jacobeva, 2024-06-06:
The makefile needs a revision. I am presently working on a feature for a different branch, as soon as that is finalised (should be done today), I will take a look at this, thanks!

starlight-traveler, 2024-06-07:
Alright! Will look forward to the update. Really interested in seeing if 2.4GHz LoRa would have a lower latency.

jacobeva, 2024-06-08:
> I do not see any documentation on how to set up a build environment
You are completely right. I plan on changing this very soon!

jacobeva, 2024-06-10:
> as soon as that is finalised (should be done today), I will take a look at this, thanks!

Yeah, it took longer than I expected, lol.

That's an ESP32-based board, right? Did you run `make prep-esp32` before compiling?

jacobeva, 2024-06-10:
I just added a build guide. Please see [here](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Documentation/BUILDING.md) for instructions. But basically, you just need to run `make prep-esp32` I think :)

starlight-traveler, 2024-06-10:
Yep, that was it simply! Guess the eyes just glossed over that argument! Thanks for the docs, will help a lot of people I am sure.

Though I have run into a linker error, which I think is related to #10. Will try and see if I can get around it (seems like its related to the newest arduino-esp32, see here: https://github.com/espressif/arduino-esp32/discussions)

`fatal error: soc/rtc_wdt.h: No such file or directory
   77 |           #include "soc/rtc_wdt.h"
      |                    ^~~~~~~~~~~~~~~
compilation terminated.

Used library    Version Path
SPI             2.0.0   /Users/foo/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/libraries/SPI
EEPROM          2.0.0   /Users/foo/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/libraries/EEPROM
BluetoothSerial 2.0.0   /Users/foo/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/libraries/BluetoothSerial
Crypto          0.4.0   /Users/foo/Documents/Arduino/libraries/Crypto

Used platform Version Path
esp32:esp32   3.0.1   /Users/foo/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1`

starlight-traveler, 2024-06-10:
Really odd, after downgrading to arduino-esp32 esp32@2.0.17, and replacing the Utilities.h reference to "hal/wdt_hal.h", made the error go away, which I see was documented. Can't realize why that would be the case but it compiles, now its time for trying to get it to work, thanks for the help!

closed 2024-06-10 by starlight-traveler

jacobeva, 2024-06-11:
> Really odd, after downgrading to arduino-esp32 ***@***.***, and replacing the Utilities.h reference to "hal/wdt_hal.h", made the error go away, which I see was documented. Can't realize why that would be the case but it compiles, now its time for trying to get it to work, thanks for the help!

Building for ESP32 is a little dodgy at the moment. This seems to be
because the BSP ends up breaking APIs on version changes, such as this
one. I assume the linker errors are because of breaking changes too,
which it seems have not been addressed by Espressif.