liberatedsystems/RNode_Firmware_CE #12: Lilygo T3S3 message "Firmware Corrupt" and not useable by reticulum

tomuk5, 2024-07-02:
After a bit of wangling with the T3S3, i finally got it flashed and the EEPROM to flash and for the rnodeconf command to query a flashes T3S3 as below:

```[16:22:03] Opening serial port /dev/ttyACM2...
[16:22:06] Device connected
[16:22:06] Current firmware version: 1.72
[16:22:06] Reading EEPROM...
[16:22:07] EEPROM checksum correct
[16:22:07] Device signature validated
[16:22:07] 
[16:22:07] Device info:
[16:22:07] 	Product            : RNode 820 - 960 MHz (03:a6:42)
[16:22:07] 	Device signature   : Validated - Local signature
[16:22:07] 	Firmware version   : 1.72
[16:22:07] 	Hardware revision  : 1
[16:22:07] 	Serial number      : 00:00:00:05
[16:22:07] 	Modem chip         : SX1262
[16:22:07] 	Frequency range    : 820.0 MHz - 1020.0 MHz
[16:22:07] 	Max TX power       : 22 dBm
[16:22:07] 	Manufactured       : 2024-07-02 16:20:40
[16:22:07] 	Device mode        : Normal (host-controlled)```

There is still a message on the device screen stating `Firmware Corrupt` that doesnt seem to go away even after power cycle or reset. (assume this is a cosmetic issue as it can be queried as above and shows as "working")

Also when i try to use the TS3 Radio with Meshchat or Reticulum directly it will loop through this error:
```[2024-07-02 16:33:36] [Notice] Opening serial port /dev/ttyACM0...
[2024-07-02 16:33:38] [Notice] Serial port /dev/ttyACM0 is now open
[2024-07-02 16:33:38] [Error] Radio state mismatch
[2024-07-02 16:33:38] [Error] After configuring RNodeInterface[RNode LoRa Interface], the reported radio parameters did not match your configuration.
[2024-07-02 16:33:38] [Error] Make sure that your hardware actually supports the parameters specified in the configuration
[2024-07-02 16:33:38] [Error] Aborting RNode startup
```
Which when i use a Lilygo T3V1.6.1 this is ok.

Parameters for the config below, iv also tried switching the tx-power to 22 and restarting reticulum but to no avail:

```
  [[RNode LoRa Interface]]
    type = RNodeInterface
    interface_enabled = True
    port = /dev/ttyACM0
    frequency = 869525000
    bandwidth = 250000
    txpower = 17
    spreadingfactor = 11
    codingrate = 5
```

im aware that the T3S3 is not ready per-se yet, but is there anything i can be doing to collect debug or any useful information to pinpoint the cause of this issue?

jacobeva, 2024-07-02:
Hello Thomas, can you tell me how you flashed this T3S3 please? Was it through compiling manually by running `make firmware-t3s3 && make upload-t3s3`? Did you follow the instructions in the terminal when doing so? It looks like the firmware hash hasn't been set properly on the board, which should be done just after uploading.

jacobeva, 2024-07-02:
Just so you know I have edited your initial comment so it is formatted better for others to read. On GitHub you have to have the ``` code tags on a new line. M*crosoft strikes again.

tomuk5, 2024-07-02:
Hi :)

Using rnodeconf and selecting option 9 and then 2

```------------------------------------------------------------------------------
                               Installer Ready

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make sure that you are satisfied with your choices.

Serial port     : /dev/ttyACM1
Device type     : RNode 820 - 960 MHz
Platform        : ESP32
Device MCU      : Espressif Systems ESP32
Firmware file   : rnode_firmware_t3s3.zip

------------------------------------------------------------------------------

Is the above correct? [y/N] y
[16:19:01] Checking firmware file availability...
[16:19:01] Retrieving latest version info from https://unsigned.io/firmware/latest/?v=2.1.3&variant=rnode_firmware_t3s3.zip
[16:19:01] The selected firmware for this board is version 1.72
[16:19:01] Using existing firmware file: rnode_firmware_t3s3.zip for version 1.72
[16:19:01] Verifying firmware integrity...
[16:19:03] 
[16:19:03] Please put the board into flashing mode now, by holding the BOOT or PRG button,
[16:19:03] while momentarily pressing the RESET button. Then release the BOOT or PRG button.
[16:19:03] Hit enter when this is done.

[16:19:13] Decompressing firmware...
[16:19:13] Firmware decompressed
[16:19:13] Flashing RNode firmware to device on /dev/ttyACM1
esptool.py v3.3.3
Serial port /dev/ttyACM1
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:54:75:f0:a8:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00010000 to 0x000f4fff...
Flash will be erased from 0x00210000 to 0x003effff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 781.2 kbit/s)...
Hash of data verified.
Compressed 15104 bytes to 10430...
Wrote 15104 bytes (10430 compressed) at 0x00000000 in 0.2 seconds (effective 494.3 kbit/s)...
Hash of data verified.
Compressed 937600 bytes to 573260...
Wrote 937600 bytes (573260 compressed) at 0x00010000 in 8.6 seconds (effective 873.3 kbit/s)...
Hash of data verified.
Compressed 1966080 bytes to 1782227...
Wrote 1966080 bytes (1782227 compressed) at 0x00210000 in 18.8 seconds (effective 835.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 570.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[16:19:46] Done flashing
[16:19:46] Waiting for ESP32 reset...
[16:19:53] 
[16:19:53] Please take the board out of flashing mode by momentarily pressing the RESET button.
[16:19:53] Hit enter when this is done.

[16:20:17] Opening serial port /dev/ttyACM1...
[16:20:20] Device connected
[16:20:20] Current firmware version: 1.72
[16:20:20] Reading EEPROM...
[16:20:21] Clearing old EEPROM, this will take about 15 seconds...
[16:20:34] Waiting for ESP32 reset...
[16:20:40] Loading signing key...
[16:20:40] 
[16:20:40] Please reset the board by momentarily pressing the RESET button.
[16:20:40] Hit enter when this is done.

[16:21:04] Bootstrapping device EEPROM...
[16:21:06] EEPROM written! Validating...
[16:21:08] 
[16:21:08] Please reset the board by momentarily pressing the RESET button.
[16:21:08] Hit enter when this is done.

[16:21:25] EEPROM Bootstrapping successful!

RNode Firmware autoinstallation complete!

To use your device with Reticulum, read the documetation at:

https://markqvist.github.io/Reticulum/manual/gettingstartedfast.html

* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *
                              Important!            

ESP32-based RNodes are created with the RNode Bootstrap Console on-board.

This repository is hosted directly on the RNode, and contains a wealth of
information, software and tools.

The RNode Bootstrap Console also contains everything needed to build
and replicate RNodes, including detailed build recipes, 3D-printable
cases, and copies of the source code for both the RNode Firmware,
Reticulum and other utilities.

To activate the RNode Bootstrap Console, power up your RNode and press
the reset button twice with a one second interval. The RNode will now
reboot into console mode, and activate a WiFi access point for you to
connect to. The console is then reachable at: http://10.0.0.1

* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *

Thank you for using this utility! Please help the project by
contributing code and reporting bugs, or by donating!

Your contributions and donations directly further the realisation
of truly open, free and resilient communications systems.

  Ethereum : 0xFDabC71AC4c0C78C95aDDDe3B4FA19d6273c5E73
  Bitcoin  : 35G9uWVzrpJJibzUwpNUQGQNFzLirhrYAH
  Monero   : 87HcDx6jRSkMQ9nPRd5K9hGGpZLn2s7vWETjMaVM5KfV4TD36NcYa8J8WSxhTSvBzzFpqDwp2fg5GX2moZ7VAP9QMZCZGET
  Ko-Fi    : https://ko-fi.com/markqvist

  Info     : https://unsigned.io/
  Code     : https://github.com/markqvist

[16:21:25] WARNING: Could not backup device EEPROM to disk
```

tomuk5, 2024-07-02:
Cheers, thats done the trick! :)

Warning message gone and functional in Reticulum

jacobeva, 2024-07-02:
Good to hear.

jacobeva, 2024-07-02:
Actually I'll keep this open as a reminder to check why the installer didn't set the hash.

jacobeva, 2024-07-09:
Could you please tell me if you specified the port when you ran the autoinstaller and it isn't set the FW hash?

tomuk5, 2024-07-09:
yes, i did specify the port when running the autoinstaller

jacobeva, 2024-07-09:
Could you try to do it again without specifying the port please? Does it set the firmware hash or not?

tomuk5, 2024-07-14:
Apologies for the delay in responding, have re-run without specifying a port.

- First atttempt, following the instructions on-screen for reset at the appropriate times, this fails just before writing the EEPROM (the port changes from /dev/ttyACM0 to /dev/ttyACM1)

- second attempt, following the instructions to reset, but instead of just using the reset button, unplug the USB cable, then press reset and then re-attach the USB cable (so the device is reset AND picks up the same port /dev/ttyACM0) this then works exactly as expected and EEPROM is written properly

jacobeva, 2024-07-14:
Thanks mate. That's really helpful. God bless.

jacobeva, 2024-07-15:
Here's what the issue was, [here](https://github.com/markqvist/Reticulum/blob/master/RNS/Utilities/rnodeconf.py#L2768) in rnodeconf setting the firmware hash is disabled if the port has been specified. No idea why this is the case. I'll fix this when I eventually fork rnodeconf, but closing this for now as it's not an issue in this repo.

closed 2024-07-15 by jacobeva