liberatedsystems/RNode_Firmware_CE #14: Project structure

jacobeva, 2024-07-16:
Generally, some parts of this firmware are not structured very well. For example, to my knowledge it is best practice for header files to only contain declarations, not definitions. In this way, the current structure of `Input.h` for example is not ideal. Such definitions should only be in cpp files ideally.

Some of the structure with regard to how these header files are imported is also somewhat questionable, and may be unnecessary if they can be eliminated altogether in places.

See [here](https://forum.arduino.cc/t/organising-code-for-a-large-scale-modular-project/591328/5).

jacobeva, 2024-08-01:
Further to this, dependency files will now be moved into a `src` directory for the project, which will improve the structure and keep the base directory tidy.

iboofkratom, 2025-09-16:
Hi
I would like to suggest that during the restructure, we also move from Arduino build system to PlatformIO. I think it would help us with following:

- PlatformIO IMO naturally leads to a better structure of the code (with /include, /lib and /src folders).
- It also has much more robust build system, dependency management, supports unit testing and debugging.
- It also supports converting Arduino projects to PlatformIO, so the transition will not be that hard. It seems to me that only thing that will take effort is converting the Makfile to platformio.ini file (to support building for all the architectures).
- We could then use  ([Portduino](https://github.com/meshtastic/framework-portduino/)) for running the RNode FW directly on linux devices like RaspberryPi (or normal computers).
- This would for example enable following setup: RPi with LoRa module connected directly to its GPIO pins. The RPi could then also run `rnsd` and use the LoRa module to communicate. (Without ESP32 or any other microcontroller.)(Meshtastic also does [this](https://meshtastic.org/docs/hardware/devices/linux-native-hardware/).) It should then be also possible to use LoRa USB dongles for Reticulum (USB-SPI converter + LoRa SPI module).
- Another benefit of Portduino is that it simplifies testing as it also allows to emulate the SPI, GPIO and other interfaces.

I am currently working on getting the RNode Firmware CE to run on linux on my RPi. Most of the stuff seems to work, but I am not able to verify it because I do not have second LoRa device.

I converted the project to PlatformIO myself, but it is very messy. If the whole project transitions from arduino-cli + Makefile to PlatformIO, I can than make a clean implementation for running RNode FW natively on linux and create pull request in this repo.

I will also gladly help with the restructuring.

Adam