liberatedsystems/RNode_Firmware_CE #18: Adding support for ESP32C3

nathmo, 2024-07-31:
Hi there,
I'm trying to compile the firmware for the ESP32-C3 and a SX1278 (my goal is to make the cheapest RNode possible)
[image]

```
1X ESP32C3 (2.5$) https://fr.aliexpress.com/item/1005006211660074.html
1X SX1278  (2.5$) https://fr.aliexpress.com/item/1005005968916380.html
TOTAL 5$
```

the wiring is rather simple : 
```
ESP32C3                             SX1278  
9, // not connected  <-X
8, // onboard led      <-X
7, // pin_ss               <---->      NSS
4, // pin_sclk            <---->      SLCK
6, // pin_mosi          <---->      MOSI
5, // pin_miso          <---->      MISO
3, // pin_reset          <---->      REST
2, // pin_dio             <---->      DIO0
3V3                          <---->      3.3V
GND                         <---->      GND
```

this is my current progress : 
https://github.com/nathmo/RNode_Firmware_CE/tree/master

I had to remove some dependency to unsigned.io which is down a lot lately for some reason.

I can build firmware, but the board is not behaving as it should (I get no answer from the rnode when I try to communicate with it using the rnodeconf tool) (the board was resetted / is not in programming mode and the serial port oppened correctly)
Also, I only get a solid LED instead of a blinking one.

If you have time, I would love to get your opinion on it (and a pointer on why the board is not answering)

If you want to test my code, look at the two pull request for the doc which should allow you to make one from scratch rather quickly if anything don't behave as expected please tell me.

nathmo, 2024-07-31:
after adding a few print statement I narrowed to this the section that cause the boot to fail

```
        if (selected_radio->preInit()) {
          modems_installed = true;
          uint32_t lfr = selected_radio->getFrequency();
          if (lfr == 0) {
            // Normal boot
          } else if (lfr == M_FRQ_R) {
            // Quick reboot
            #if HAS_CONSOLE
              if (rtc_get_reset_reason(0) == POWERON_RESET) {
                console_active = true;
              }
            #endif
          } else {
            // Unknown boot
          }
          selected_radio->setFrequency(M_FRQ_S);
        } else {
          modems_installed = false;
        }
        if (!modems_installed) {
            break;
        }
    }
```

nathmo, 2024-07-31:
Oh and before I forget there was another strange thing : 
I had to hardcode the wifi SSID because it used the one set by the Bluetooth and if the board doesnt have bluetooth the compilation crash. idk it this was a known issue

jacobeva, 2024-07-31:
> Oh and before I forget there was another strange thing : I had to hardcode the wifi SSID because it used the one set by the Bluetooth and if the board doesnt have bluetooth the compilation crash. idk it this was a known issue

I have never looked at the console functionality to be honest, I've barely used it. That said, this sounds kind of weird indeed.

jacobeva, 2024-07-31:
Have you tried using a tool such as `picocom` to read the serial port your RNode is connected to and take a look at what may be causing the crash? Unlike the nRF52, the ESP32 usually spits out an error if it crashes.

jacobeva, 2024-07-31:
Try changing [this line as seen in my commit to a test branch](https://github.com/liberatedsystems/RNode_Firmware_CE/commit/f328e4ce4ad90d0f94777e5a69f771c27faba11d) to see if it will help at all. Currently a few ESP32 targets are experiencing crashes like you are describing as far as I know. If that fixes it, then it proves it is probably stack issues across the board (as I expect).

nathmo, 2024-07-31:
> Have you tried using a tool such as `picocom` to read the serial port your RNode is connected to and take a look at what may be causing the crash? Unlike the nRF52, the ESP32 usually spits out an error if it crashes.

Thats how I narrowed it to the section of code mentionnes previoulsy but I have to dig deeper on the why.
Also there was not crash reason. It just freeze between two print statement

nathmo, 2024-07-31:
> I have never looked at the console functionality to be honest, I've barely used it. That said, this sounds kind of weird indeed.

Console functionnality ? I'm not sure what you are refering to.

I meant that I had to modify the code because there is a wifi hotspot functionnality (dont ask me why) and the ssid was set according to the bluettooth but since the code is modular and some boars dont have bluetooth the compilation crash for theses boards. 
This feels like a bug that might appear later on on another board that don't have bluetooth.

jacobeva, 2024-08-01:
> Console functionnality ? I'm not sure what you are refering to.
>
> I meant that I had to modify the code because there is a wifi hotspot functionnality (dont ask me why) and the ssid was set according [...]

Each RNode comes with the RNode Console, which hosts the source code for the repo and various other related Reticulum stuff. It does this over a wifi hotspot. This is what you had to change to get it to work.

jacobeva, 2024-08-01:
Did my fix on that branch work?

jacobeva, 2024-08-01:
If my reckoning is correct, it will.

nathmo, 2024-08-03:
I did, and sadly it's not the issue that's blocking the boot, :( thanks for the help tho :) .
currently, I traced the issue to the following : 
1) in the RNode_Firmware_CE.ino : Line 230 `if (selected_radio->preInit())` 
2) in Radio.cpp : Line 1138 of sx127x::preInit() `version = readRegister(REG_VERSION_7X);`
3) in Radio.cpp : Line 1098 of sx127x::readRegister(uint8_t address) { return singleTransfer(address & 0x7f, 0x00); }
4) in Radio.cpp : Line 1158 of  sx127x::singleTransfer : _spiModem.beginTransaction(_spiSettings);

_spiModem.begin() was called
https://www.arduino.cc/reference/en/language/functions/communication/spi/begintransaction/

Currently looking online if there is anything I missed regarding the ESP32C3 that would explain this behaviour

nathmo, 2024-08-03:
this issue gave me the hint :) 
https://github.com/adafruit/Adafruit-PN532/issues/53

and this commit was the answer : https://github.com/nathmo/RNode_Firmware_CE/commit/2217fe07714017086a672b2e19b34d6271e378c3

the node seems to work. I will do more testing.

(I tried to set the stack to its original size and the node seems to behave normally. the led is steadily blinking and I can run `` rnodeconf /dev/ttyACM0 -i`

nathmo, 2024-08-03:
btw how do I sign the firmware ? with unsigned.io down I can't find how to do it 

I guess there should be somehere that I can configure the EEPROM so that it shows the correct info but I can't quite get what to do.

``` rnodeconf /dev/ttyACM0 -i
[18:13:00] Opening serial port /dev/ttyACM0...
[18:13:03] Device connected
[18:13:03] Current firmware version: 1.73
[18:13:03] Reading EEPROM...
[18:13:04] EEPROM checksum correct
[18:13:04] Device signature validation failed

     WARNING! This device is NOT verifiable and should NOT be trusted.
     Someone could have added privacy-breaking or malicious code to it.

     Please verify the signing key is present on this machine.
     Autogenerated keys will not match another machine's signature.

     Proceed at your own risk and responsibility! If you created this
     device yourself, please read the documentation on how to sign your
     device to avoid this warning.

     Always use a firmware downloaded as binaries or compiled from source
     from one of the following locations:

        https://unsigned.io/rnode
        https://github.com/markqvist/rnode_firmware

     You can reflash and bootstrap this device to a verifiable state
     by using this utility. It is recommended to do so NOW!

     To initialise this device to a verifiable state, please run:

              rnodeconf /dev/ttyACM0 --autoinstall

[18:13:04]
[18:13:04] Device info:
[18:13:04]      Product            : RNode (Band capabilities unknown) (03:ff:3b)
[18:13:04]      Device signature   : Unverified
[18:13:04]      Firmware version   : 1.73
[18:13:04]      Hardware revision  : 1
[18:13:04]      Serial number      : 00:00:00:07
[18:13:04]      Modem chip         : Unknown
[18:13:04]      Frequency range    : 100.0 MHz - 1100.0 MHz
[18:13:04]      Max TX power       : 14 dBm
[18:13:04]      Manufactured       : 2024-07-20 15:48:49
[18:13:04]      Device mode        : Normal (host-controlled)```

jacobeva, 2024-08-03:
> I did, and sadly it's not the issue that's blocking the boot, :( thanks for the help tho :) . currently, I traced the issue to the following :
>
>     1. in the RNode_Firmware_CE.ino : Line 230 `if [...]

I had a similar issue before, I know it sounds really stupid, but trust me, it's a stack issue.

I had a freeze when the modem was being shut down via SPI before, tested it exactly the way you did. Don't trust your findings, they are a red herring. The stack is too small, (or something similar with memory) this is your actual problem.

I will see if there is any way to make the stack larger in the setup routine for freeRTOS.

jacobeva, 2024-08-03:
Oh I am sorry, I didn't read your second comment, d'oh! Glad you fixed it!!

nathmo, 2024-08-03:
no problem :) I'm currently trying to figure out how to connect my two rasperry pi over reticulum using the two Rnode I just made

The device firmware dont seems to matter as it just print a huge warning (or I missed something)
I tried to make a release of the firmware using `make release-genericesp32c3` I have a zip but the rnodeconf is not super clear on how to flash from a zip that is avaible locally.

jacobeva, 2024-08-03:
> btw how do I sign the firmware ? with unsigned.io down I can't find how to do it
> 
> I guess there should be somehere that I can configure the EEPROM so that it shows the correct info but I can't quite get what to do.
> 
> ```
> [18:13:00] Opening serial port /dev/ttyACM0...
> [18:13:03] Device connected
> [18:13:03] Current firmware version: 1.73
> [18:13:03] Reading EEPROM...
> [18:13:04] EEPROM checksum correct
> [18:13:04] Device signature validation failed
> 
>      WARNING! This device is NOT verifiable and should NOT be trusted.
>      Someone could have added privacy-breaking or malicious code to it.
> 
>      Please verify the signing key is present on this machine.
>      Autogenerated keys will not match another machine's signature.
> 
>      Proceed at your own risk and responsibility! If you created this
>      device yourself, please read the documentation on how to sign your
>      device to avoid this warning.
> 
>      Always use a firmware downloaded as binaries or compiled from source
>      from one of the following locations:
> 
>         https://unsigned.io/rnode
>         https://github.com/markqvist/rnode_firmware
> 
>      You can reflash and bootstrap this device to a verifiable state
>      by using this utility. It is recommended to do so NOW!
> 
>      To initialise this device to a verifiable state, please run:
> 
>               rnodeconf /dev/ttyACM0 --autoinstall
> 
> [18:13:04]
> [18:13:04] Device info:
> [18:13:04]      Product            : RNode (Band capabilities unknown) (03:ff:3b)
> [18:13:04]      Device signature   : Unverified
> [18:13:04]      Firmware version   : 1.73
> [18:13:04]      Hardware revision  : 1
> [18:13:04]      Serial number      : 00:00:00:07
> [18:13:04]      Modem chip         : Unknown
> [18:13:04]      Frequency range    : 100.0 MHz - 1100.0 MHz
> [18:13:04]      Max TX power       : 14 dBm
> [18:13:04]      Manufactured       : 2024-07-20 15:48:49
> [18:13:04]      Device mode        : Normal (host-controlled)```
> ```

When you provisioned the EERPOM, which device did you use to do so? Did you use the device you are using to connect with it now?

To explain, a key is generated on the machine using rnodeconf when the EEPROM is provisioned, and there is a signature written to the EEPROM which is created from some of the info in the EEPROM and the key. Because this is a locally generated key, it will be fine when connected to that machine (i.e. no errors), but with others this error will show.

To fix the error, do as follows:
Run `rnodeconf -p` on the device you used to provision the EEPROM
Copy the EEPROM key it spits out, and run the following on the device you wish to use the RNode with:
`rnodeconf --trust-key [paste key here]`

Should fix the error.

jacobeva, 2024-08-03:
Only one key is currently trusted by default in `rnodeconf`, and that is Mark's (the creator of Reticulum) signing key. Nothing else.

jacobeva, 2024-08-03:
If you are able to get communication working between these nodes, I will be surprised. Please let me know if it does work, if that's the case I'll need to change the code a bit.

nathmo, 2024-08-03:
note 1) I realise there is some delay in github message :') 

now about the EEPROM provisioning, I tried to do it once (but it was three weeks ago and on another computer) 

I just ran : ``rnodeconf /dev/ttyACM0 --eeprom-wipe` from my raspberry pi from which the rnode is connected.
the I tried `rnodeconf -r /dev/ttyACM0`
but I get 
```
 rnodeconf -r /dev/ttyACM0
[19:23:36] Opening serial port /dev/ttyACM0...
[19:23:40] Device connected
[19:23:40] Current firmware version: 1.73
[19:23:40] Reading EEPROM...
[19:23:40] Waiting for ESP32 reset...
Traceback (most recent call last):
  File "/home/ubuntu/RNode_Firmware_CE/venv/bin/rnodeconf", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/ubuntu/RNode_Firmware_CE/venv/lib/python3.11/site-packages/RNS/Utilities/rnodeconf.py", line 3310, in main
    if len(args.model) == 2:
       ^^^^^^^^^^^^^^^
TypeError: object of type 'NoneType' has no len()
```

I vaguelly recall that the first time I provisionned the EEPROM I had to read the python script of rnodeconf to understand what argugment where required and just tried with a random value. But I dont remeber seeing anywhere a full example of the syntax of the command to provision the EEPROM and what parameters influence what. (   --platform platform   Platform specification for device bootstrap
  --product product     Product specification for device bootstrap
  --model model         Model code for device bootstrap
  --hwrev revision      Hardware revision for device bootstrap)

like where do I set the radio I have on board ? like power and frequency ?

nathmo, 2024-08-03:
it's comming back to me now I think.

```models = {
    0x11: [430000000, 510000000, 22, "430 - 510 MHz", "rnode_firmware_rak4631.zip", "SX1262"],
    0x12: [779000000, 928000000, 22, "779 - 928 MHz", "rnode_firmware_rak4631.zip", "SX1262"],
    0xA4: [410000000, 525000000, 14, "410 - 525 MHz", "rnode_firmware.hex", "SX1278"],
    0xA9: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware.hex", "SX1276"],
    0xA1: [410000000, 525000000, 22, "410 - 525 MHz", "rnode_firmware_t3s3.zip", "SX1268"],
    0xA6: [820000000, 1020000000, 22, "820 - 960 MHz", "rnode_firmware_t3s3.zip", "SX1262"],
    0xA2: [410000000, 525000000, 17, "410 - 525 MHz", "rnode_firmware_ng21.zip", "SX1278"],
    0xA7: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware_ng21.zip", "SX1276"],
    0xA3: [410000000, 525000000, 17, "410 - 525 MHz", "rnode_firmware_ng20.zip", "SX1278"],
    0xA8: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware_ng20.zip", "SX1276"],
    0xB3: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v20.zip", "SX1278"],
    0xB8: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v20.zip", "SX1276"],
    0xB4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v21.zip", "SX1278"],
    0xB9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v21.zip", "SX1276"],
    0x04: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v21_tcxo.zip", "SX1278"],
    0x09: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v21_tcxo.zip", "SX1276"],
    0xBA: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v10.zip", "SX1278"],
    0xBB: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v10.zip", "SX1276"],
    0xC4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_heltec32v2.zip", "SX1278"],
    0xC9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_heltec32v2.zip", "SX1276"],
    0xC5: [470000000, 510000000, 21, "470 - 510 MHz", "rnode_firmware_heltec32v3.zip", "SX1262"],
    0xCA: [863000000, 928000000, 21, "863 - 928 MHz", "rnode_firmware_heltec32v3.zip", "SX1262"],
    0xE4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_tbeam.zip", "SX1278"],
    0xE9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_tbeam.zip", "SX1276"],
    0xE3: [420000000, 520000000, 22, "420 - 520 MHz", "rnode_firmware_tbeam_sx1262.zip", "SX1268"],
    0xE8: [850000000, 950000000, 22, "850 - 950 MHz", "rnode_firmware_tbeam_sx1262.zip", "SX1262"],
    0xFE: [100000000, 1100000000, 17, "(Band capabilities unknown)", None, "Unknown"],
    0xFF: [100000000, 1100000000, 14, "(Band capabilities unknown)", None, "Unknown"],
}```

theses are the model code that represent the radio and hardware

and last time I set it as unknown due to the inexistance of my board there.

It's more something for the RNS side of things but it might be a good idea to separate the radio parameters from the board itself. especially if each new board require a pull request to the rnodeconf utility

jacobeva, 2024-08-03:
> It's more something for the RNS side of things but it might be a good idea to separate the radio parameters from the board itself. especially if each new board require a pull request to the rnodeconf utility
`rnodeconf` is on my list to overhaul, do not worry.

jacobeva, 2024-08-03:
> --product product Product specification for device bootstrap
> --model model Model code for device bootstrap
> --hwrev revision Hardware revision for device bootstrap)

These are the only things you need to set to provision the EEPROM. 

Have you managed to communicate between the two boards successfully? :)

nathmo, 2024-08-03:
Well if you need a hand I will be glad to assist.
I might make a pull request with more documentation once I ironned out the king of flashing a board from A to Z.

jacobeva, 2024-08-03:
Haha. Please do feel free, especially some instructions on provisioning with `rnodeconf` would be helpful my friend.

nathmo, 2024-08-03:
> These are the only things you need to set to provision the EEPROM.
>
> Have you managed to communicate between the two boards successfully? :)

no not yet I'm still rereading the device code.

trying to make this right. (what make you think I should have an issue making the two board communicating ?)
but dont worry If I manage to send a packet between the two I will tell you :)

jacobeva, 2024-08-03:
I have a feeling your change will mean the modem isn't connected over SPI correctly, but I could be wrong. Given your RNode isn't complaining that it's not installed, it probably has managed to read the version, so it might be alright. Time will tell.

nathmo, 2024-08-03:
okay I think I got it so for anyone that stumble upon this before a better documentation is written : 

1) generate a new signing key
`rnodeconf -k`

which return something like this
```
[20:54:28] Generating a new device signing key...
[20:54:28] Device signing key written to /home/ubuntu/.config/rnodeconf/firmware/device.key
[20:54:28] Generating a new EEPROM signing key...
[20:54:28] Wrote signing key
[20:54:28] Public key:
[20:54:28] 30819f300d06092a864886f70d010101050003818d0030818902818100cb64a5c11a0b82bf9475f67e8ce4f01887165b5328a55e01e9022bf32f419152ffdb5292ea3c6ab3d3d064f28a91feabea87e89244788898928d1d0abbd3ac91d91a8ac2a57a3979fac6e3d7f06ab92f3e4a3a60cc5b34094f6d26249e89e6737717f13d7a5520b9eceb2bed2b36d19cdddf3b9a6649a1ca50dc73cab44d49a50203010001
```

2) compile the firmware
` make firmware-genericesp32c3`

3) flash the board
`make upload-genericesp32c3`

4) erase the eeprom
` rnodeconf /dev/ttyACM0 --eeprom-wipe`

5) you provision the eeprom
`rnodeconf /dev/ttyACM1 -r --platform ESP32 --model a4 --product f0 --hwrev 3`

--hwrev is required (any integer between 1 and 255 should work)
--platform must either be AVR ESP32 or NRF52 
--model should be something from the list at the end of this message (11 or a4 or b8) (no caps, no 0x)
--product should also be a 2 char code from the following mapping : 

```
PRODUCT_RAK4631 = 0x10
PRODUCT_RNODE  = 0x03
PRODUCT_T32_10 = 0xB2
PRODUCT_T32_20 = 0xB0
PRODUCT_T32_21 = 0xB1
PRODUCT_H32_V2 = 0xC0
PRODUCT_H32_V3 = 0xC1
PRODUCT_TBEAM  = 0xE0
PRODUCT_HMBRW  = 0xF0
```
and once you run the command you should have something like this : 

```
[20:51:08] Opening serial port /dev/ttyACM1...
[20:51:11] Device connected
[20:51:11] Current firmware version: 1.73
[20:51:11] Reading EEPROM...
[20:51:12] Waiting for ESP32 reset...
[20:51:18] Loading signing key...
```

currently that last command will crash due to 
https://github.com/markqvist/Reticulum/pull/512
not being merged :( 
but you get the idea  :) 
(and you can always clone repo, apply the patch locally and do it)

the said list : 
```
models = {
    0x11: [430000000, 510000000, 22, "430 - 510 MHz", "rnode_firmware_rak4631.zip", "SX1262"],
    0x12: [779000000, 928000000, 22, "779 - 928 MHz", "rnode_firmware_rak4631.zip", "SX1262"],
    0xA4: [410000000, 525000000, 14, "410 - 525 MHz", "rnode_firmware.hex", "SX1278"],
    0xA9: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware.hex", "SX1276"],
    0xA1: [410000000, 525000000, 22, "410 - 525 MHz", "rnode_firmware_t3s3.zip", "SX1268"],
    0xA6: [820000000, 1020000000, 22, "820 - 960 MHz", "rnode_firmware_t3s3.zip", "SX1262"],
    0xA2: [410000000, 525000000, 17, "410 - 525 MHz", "rnode_firmware_ng21.zip", "SX1278"],
    0xA7: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware_ng21.zip", "SX1276"],
    0xA3: [410000000, 525000000, 17, "410 - 525 MHz", "rnode_firmware_ng20.zip", "SX1278"],
    0xA8: [820000000, 1020000000, 17, "820 - 1020 MHz", "rnode_firmware_ng20.zip", "SX1276"],
    0xB3: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v20.zip", "SX1278"],
    0xB8: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v20.zip", "SX1276"],
    0xB4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v21.zip", "SX1278"],
    0xB9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v21.zip", "SX1276"],
    0x04: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v21_tcxo.zip", "SX1278"],
    0x09: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v21_tcxo.zip", "SX1276"],
    0xBA: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_lora32v10.zip", "SX1278"],
    0xBB: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_lora32v10.zip", "SX1276"],
    0xC4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_heltec32v2.zip", "SX1278"],
    0xC9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_heltec32v2.zip", "SX1276"],
    0xC5: [470000000, 510000000, 21, "470 - 510 MHz", "rnode_firmware_heltec32v3.zip", "SX1262"],
    0xCA: [863000000, 928000000, 21, "863 - 928 MHz", "rnode_firmware_heltec32v3.zip", "SX1262"],
    0xE4: [420000000, 520000000, 17, "420 - 520 MHz", "rnode_firmware_tbeam.zip", "SX1278"],
    0xE9: [850000000, 950000000, 17, "850 - 950 MHz", "rnode_firmware_tbeam.zip", "SX1276"],
    0xE3: [420000000, 520000000, 22, "420 - 520 MHz", "rnode_firmware_tbeam_sx1262.zip", "SX1268"],
    0xE8: [850000000, 950000000, 22, "850 - 950 MHz", "rnode_firmware_tbeam_sx1262.zip", "SX1262"],
    0xFE: [100000000, 1100000000, 17, "(Band capabilities unknown)", None, "Unknown"],
    0xFF: [100000000, 1100000000, 14, "(Band capabilities unknown)", None, "Unknown"],
}
```

nathmo, 2024-08-03:
@jacobeva so I tried and there is an issue.
I tried to set the stack to 15KB but the issue persists

when I try to run rnsd I get an error that the software was unable to configure the radio
```
[2024-08-03 22:56:36] [Verbose] Attempting to reconnect serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_34:B7:DA:F8:7C:3C-if00 for RNodeInterface[RNode LoRa Interface]...
[2024-08-03 22:56:36] [Notice] Opening serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_34:B7:DA:F8:7C:3C-if00...
[2024-08-03 22:56:38] [Notice] Serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_34:B7:DA:F8:7C:3C-if00 is now open
[2024-08-03 22:56:38] [Verbose] Configuring RNode interface...
[2024-08-03 22:56:38] [Verbose] Waiting for radio configuration validation for RNodeInterface[RNode LoRa Interface]...
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting frequency is 0.0 MHz
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting bandwidth is 7.8 KHz
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting TX power is 0 dBm
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting spreading factor is 8
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting coding rate is 5
[2024-08-03 22:56:38] [Verbose] RNodeInterface[RNode LoRa Interface] On-air bitrate is now 0.19 kbps
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting state is offline
[2024-08-03 22:56:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting state is offline
[2024-08-03 22:56:39] [Error] Frequency mismatch
[2024-08-03 22:56:39] [Error] Bandwidth mismatch
[2024-08-03 22:56:39] [Error] TX power mismatch
[2024-08-03 22:56:39] [Error] Radio state mismatch
[2024-08-03 22:56:39] [Error] After configuring RNodeInterface[RNode LoRa Interface], the reported radio parameters did not match your configuration.
[2024-08-03 22:56:39] [Error] Make sure that your hardware actually supports the parameters specified in the configuration
[2024-08-03 22:56:39] [Error] Aborting RNode startup
[2024-08-03 22:56:43] [Error] An error occurred while detaching RNodeInterface[RNode LoRa Interface]. The contained exception was: Attempting to use a port that is not open
```
then I resetted the SPI line that I changed but now the device freeze and the RNS deamon cant open the serial port anymore

Do you have any other idea ?

jacobeva, 2024-08-03:
Ah, I forgot you don't have a display, so you can't tell if the modem init failed. Well, this is what I expected to happen. You can't use the RNode because it's not connected to the modem's SPI properly.

I'll have to take a closer look at that issue you linked and see how I may fix it.

nathmo, 2024-08-03:
I might have one thoses I2C display. I could try to modify the firmware to support it. it might help the debbugging effort but it will take a few days

jacobeva, 2024-08-03:
Don't bother, just open the serial port quickly after power up to see the boot message :) It'll probably say something like "no installed modem found".

nathmo, 2024-08-03:
[image]

this is what I get after each reboot (I rebooted 4 time in that screenshot)

jacobeva, 2024-08-03:
Change baud rate to 115200.

nathmo, 2024-08-04:
After realising that due the ESP32-C3 could not detect reset the same was as the ESP32, it could also not enter the console.
So I tried disabling it and now I get a useful crashdump :)

So my hunch was correct and it is due to a semaphore on the SPI bus

This is progress

```
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x403823a0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
pre init before begin                                        // my debug print statement
_spiModem.begin(_sclk, _miso, _mosi, _ss)      // my debug print statement
pre init after begin                                          // my debug print statement
A                                                                      // my debug print statement

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)
Core  0 register dump:
MEPC    : 0x4038279a  RA      : 0x4038a5be  SP      : 0x3fca4d00  GP      : 0x3fc91400
TP      : 0x3fc7f054  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fca4e7a  S1      : 0x0000007d  A0      : 0x3fca4d64  A1      : 0x3fc930f9
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc98000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca4d58  S3      : 0x00000001
S4      : 0x3fca4d58  S5      : 0x4038b25c  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fca4d00: 0x6e6d6c6b 0x6a696867 0x66656463 0x4038f74c 0x3fc9f1a4 0x3fca0f94 0x000001c0 0x3fc930f8
3fca4d20: 0x00000020 0x00000017 0x00000000 0x39343531 0xa5a5a500 0x3fc931e0 0x3c0c78d4 0x3fc935bc
3fca4d40: 0x3c0c797d 0x3fc931f0 0x3fca4d2c 0x3fc931f4 0x3c0c7d18 0x3fc930f8 0x00000000 0x00000000
3fca4d60: 0xff000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551 0x6d655365 0x6f687061
3fca4d80: 0x61546572 0x7120656b 0x65756575 0x313a632e 0x20393435 0x51787028 0x65756575 0x78753e2d
3fca4da0: 0x6d657449 0x657a6953 0x203d3d20 0x00002930 0x00000002 0x3fc96000 0x00000001 0x00000064
3fca4dc0: 0x00000003 0x00000004 0x600c2000 0x600c2000 0x00000000 0x00000000 0x3fc9f490 0x4038ad0e
3fca4de0: 0x00000000 0x00000042 0x00000000 0x4038c632 0x3fc977f0 0x00000000 0x00000000 0x00000000
3fca4e00: 0x3c0c0514 0x3fc9f43c 0x3fc9f39c 0x4038a80a 0x00000000 0x00000000 0x3fc9f39c 0x4038ad28
3fca4e20: 0x00000000 0x00000017 0x00000000 0x48587a27 0x00000000 0x00000042 0x00000000 0x000012c8
3fca4e40: 0x00000001 0x007a1200 0x3fc9d7c8 0x4038b25c 0x60043000 0x00000003 0x00000000 0xffffffff
3fca4e60: 0x3fc90c40 0x3fc977f8 0x00000017 0x3fc977e0 0x60043000 0x00000017 0x00000000 0x000012c8
3fca4e80: 0x00000001 0x007a1200 0x3fca9418 0x42005bb8 0x00000000 0x00000042 0x00000000 0x30b80001
3fca4ea0: 0x3fca9150 0x3fc96000 0x3fca9418 0x403803ba 0x00000000 0x00000000 0x00000012 0x3fca8150
3fca4ec0: 0x000001f3 0x3fc96000 0x00001432 0x42004bc2 0x00000005 0x3fc94200 0x00000000 0x00000000
3fca4ee0: 0x00002000 0x3fc977d0 0x3fca8150 0x420029c4 0x00000002 0xffffffff 0x00000000 0x00000000
3fca4f00: 0x00000000 0x00000000 0xffffff00 0x000000ff 0x00000000 0x000f4240 0x00000000 0x3fc9d7c8
3fca4f20: 0x00000000 0x3fc98000 0x00000000 0x4200a0a6 0x00000000 0x00000000 0x00000000 0x4038cc4e
3fca4f40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca4f60: 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x3fca4c90 0x0000138c
3fca4f80: 0x3fc96c1c 0x3fc96c1c 0x3fca4f78 0x3fc96c14 0x00000018 0x10d890ce 0xd1fb350e 0x3fca4f78
3fca4fa0: 0x00000000 0x00000001 0x3fca1368 0x706f6f6c 0x6b736154 0x4dcc4500 0x002526d0 0x00000000
3fca4fc0: 0x3fca4f60 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9d064 0x3fc9d0cc
3fca4fe0: 0x3fc9d134 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4209fc76
3fca5000: 0x00000000 0x00000000 0x00000000 0x3fc9f374 0x00000000 0x00000000 0x00000000 0x00000000
3fca5020: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5040: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5060: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5080: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca50a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca50c0: 0x00000000 0x00000000 0xad000000 0xbaad5678 0x00001860 0xabba1234 0x00001854 0x3fca5130
3fca50e0: 0x3fca5130 0x3fca6930 0x3fca692f 0x00000000 0x3fca50f4 0xffffffff 0x3fca50f4 0x3fca50f4

ELF file SHA256: 0426eedcb4542d9f

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x403823a0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
```

nathmo, 2024-08-04:
btw, this is for later but to fix the RNode console this might be a good start : https://github.com/espressif/arduino-esp32/issues/8040

nathmo, 2024-08-04:
I think I might have a theory :
the crash happen due to a semaphore on the SPI bus.
What lock that semaphoreis that SPI object is passed by value (copy) and not by reference.
This creates a new SPI object, but it fails when I want to use it because the other one still exist and hold the lock.

sx127x::sx127x(uint8_t index, SPIClass spi,...
sx127x::sx127x(uint8_t index, SPIClass& spi, ...

I will now try to validate that theory by modifying the class and see if it fix it

nathmo, 2024-08-04:
Well it allows the board to boot but I must have broken the radio initialisation as I can't change the radio parameters.

@jacobeva if you have an insight, I would be happy to hear it

jacobeva, 2024-08-05:
```
This creates a new SPI object, but it fails when I want to use it because the other one still exist and hold the lock.

sx127x::sx127x(uint8_t index, SPIClass spi,...
sx127x::sx127x(uint8_t index, SPIClass& spi, ...

I will now try to validate that theory by modifying the class and see if it fix it
```
You're probably right about this. I am currently working on another feature, but will investigate this issue ASAP once I am available.

nathmo, 2024-08-05:
no stress. This is my vacation project and as much as it would give me some happy juice to make it works there are no stakes on that project. (plus I'm also working on it albeit slowly)

jacobeva, 2024-08-06:
An update: I am indeed retarded. Long story short, I had the exact same issue are you are currently having, but I also had memory corruption issues at the same time, which somehow trumped that and made things run (seemingly) okay, until I began adding extra features in the branch I am working on. I misinterpreted the semaphore issue (this was like a month ago now) to be a memory issue (which was also occurring), and thought it was fixed. In actual fact, it has been here the whole time still, hence your experience. 

So, I will roll out a fix today to allow for passing a pointer into the modem class when initialising it. This should resolve the issue and restore operation to your board (and probably a lot of other ESP32 boards as well).

nathmo, 2024-08-06:
Well for your defense this is a nasty bug.
It took me quite a lot of staring at the code and really thorough testing through print statement to figure it out.
Also, some dusting of the worst kind of trick question they used my CS exam.

Anyway I'm glad that it will actually solve more than one issue at once ( im trying to understand what the code do in details because I feel like having a basic programm with only the core functionnality would be a great plus, and being able to document that in a nice document would be even better because as of know, what happen between my terminal and the RF signal is pur magic to me :)

jacobeva, 2024-08-06:
It is a nasty bug, indeed.

This is offtopic, but as you probably know by now this codebase isn't originally mine. I know it relatively well, but completing issue #1 and drawing up a document describing operation as you say, would be a good idea indeed.

jacobeva, 2024-08-06:
Commit 721cee3 should fix this SPI semaphore issue. Please copy your board config into that build, then test the updated firmware. It should fix the issue.

nathmo, 2024-08-06:
I would love to do it but it's going to take me quite a while to catch up all you had to learn the hard way.
So if you don't mind trying to make a draft where you lay out all that you know in whatever order it come so it's much easier to add and continue than starting from scratch.

Like what the RNode excpet as command, how does it accedt tge packet to transmit, what kind of processing does it do ? How does it deal with packet segmentation ? Is the Lora board running in "continuous" mode (where you can send a datastream as long as you want) or is in packet mode (I saw that in the modem datasheet but had no time to dig deeper)

But remeber don't overwork yourself either it's an open source project your not paid it should be fun / interesting

And we don't know much about each other skill level but I don't mind trying something if you can define it clearly :)

nathmo, 2024-08-07:
well, I'm having the same result as when I tried to pass it by reference :( 

The RNode boot, seems stable and answer command, but the radio is not doing what it should.

this is the reticulum config
```
[[RNode LoRa Interface]]
  type = RNodeInterface
  interface_enabled = True
  port = /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00
  frequency = 434000000
  bandwidth = 125000
  txpower = 7
  spreadingfactor = 8
```
and this is the message when trying to run rnsd
```
[2024-08-07 14:57:36] [Verbose] Attempting to reconnect serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00 for RNodeInterface[RNode LoRa Interface]...
[2024-08-07 14:57:36] [Notice] Opening serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00...
[2024-08-07 14:57:38] [Notice] Serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00 is now open
[2024-08-07 14:57:38] [Verbose] Configuring RNode interface...
[2024-08-07 14:57:38] [Verbose] Waiting for radio configuration validation for RNodeInterface[RNode LoRa Interface]...
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting frequency is 0.0 MHz
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting bandwidth is 7.8 KHz
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting TX power is 0 dBm
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting spreading factor is 8
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting coding rate is 5
[2024-08-07 14:57:38] [Verbose] RNodeInterface[RNode LoRa Interface] On-air bitrate is now 0.19 kbps
[2024-08-07 14:57:38] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting state is offline
[2024-08-07 14:57:39] [Error] Frequency mismatch
[2024-08-07 14:57:39] [Error] Bandwidth mismatch
[2024-08-07 14:57:39] [Error] TX power mismatch
[2024-08-07 14:57:39] [Error] Radio state mismatch
[2024-08-07 14:57:39] [Error] After configuring RNodeInterface[RNode LoRa Interface], the reported radio parameters did not match your configuration.
[2024-08-07 14:57:39] [Error] Make sure that your hardware actually supports the parameters specified in the configuration
[2024-08-07 14:57:39] [Error] Aborting RNode startup
```

but otherwise the node answer 
```
rnodeconf /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00 -i
[14:53:43] Opening serial port /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_F0:F5:BD:FD:BA:F8-if00...
[14:53:46] Device connected
[14:53:46] Current firmware version: 1.73
[14:53:46] Reading EEPROM...
[14:53:47] EEPROM checksum correct
[14:53:47] Device signature validated
[14:53:47]
[14:53:47] Device info:
[14:53:47]      Product            : Hombrew RNode 410 - 525 MHz (f0:a4:3b)
[14:53:47]      Device signature   : Validated - Local signature
[14:53:47]      Firmware version   : 1.73
[14:53:47]      Hardware revision  : 3
[14:53:47]      Serial number      : 00:00:00:02
[14:53:47]      Modem chip         : SX1278
[14:53:47]      Frequency range    : 410.0 MHz - 525.0 MHz
[14:53:47]      Max TX power       : 14 dBm
[14:53:47]      Manufactured       : 2024-08-03 21:46:39
[14:53:47]      Device mode        : Normal (host-controlled)
```
(and there is nothing comming out of the serial port)
[image]

jacobeva, 2024-08-09:
> Radio reporting frequency is 0.0 MHz
This isn't right. Seems the frequency isn't being set properly.

Have you made any progress with this at all? There is definitely not nothing coming out of the serial port. Reset the board and check again. If it doesn't print "No valid radio module found", then the modem is being initialised correctly.

jacobeva, 2024-08-09:
I can see [default_spi is set to true](https://github.com/nathmo/RNode_Firmware_CE/blob/master/Boards.h#L641). Try setting it to false and see if that makes any difference for you.

nathmo, 2024-08-09:
I will try that. My last test was to use print statement to manually set the frequency 

and it works I can set the value. (but I dont remeber if the value was correct of not there might have need a factor that was wrong like the set frequency didnt apply the conversion from Hz to binary for the SX1278 correctly. I need to check again.

will do in the next few days :)

```

if (selected_radio->preInit()) {
          modems_installed = true;
          uint32_t lfr = selected_radio->getFrequency();
          Serial.print("getFrequency() : "); //debug statement !
          Serial.printf("%u", lfr);    //debug statement !

          Serial.print("M_FRQ_S : "); //debug statement !
          Serial.printf("%u", M_FRQ_S);    //debug statement !

          if (lfr == 0) {
            // Normal boot
          } else if (lfr == M_FRQ_R) {
            // Quick reboot
            #if HAS_CONSOLE
              if (rtc_get_reset_reason(0) == POWERON_RESET) {
                console_active = true;
              }
            #endif
          } else {
            // Unknown boot
          }
          selected_radio->setFrequency(433000000);
          Serial.print("getFrequency() 433000000 : ");          //debug statement !
          Serial.printf("%u", selected_radio->getFrequency());    //debug statement !

        } else {
          modems_installed = false;
        }
        if (!modems_installed) {
            break;
        }

```

nathmo, 2024-08-11:
to comment on the previous issue, my  test works there is not conversion error. I can set the frequency and read it inside the programm.

I'm trying to emulate the commande sent by the rnsd to change the frequency using a serial consol that send data as hex but the lack of documentation on the "API" make me wanna scream into the void and set my computer on fire while the universe laughs at my misery.

the unducommented misery : 
```

void serialCallback(uint8_t sbyte) {
  if (IN_FRAME && sbyte == FEND && 
            (command == CMD_INT0_DATA
          || command == CMD_INT1_DATA
          || command == CMD_INT2_DATA
          || command == CMD_INT3_DATA
          || command == CMD_INT4_DATA
          || command == CMD_INT5_DATA
          || command == CMD_INT6_DATA
          || command == CMD_INT7_DATA
          || command == CMD_INT8_DATA
          || command == CMD_INT9_DATA
          || command == CMD_INT10_DATA 
          || command == CMD_INT11_DATA)) {
    IN_FRAME = false;

    if (getInterfaceIndex(command) < INTERFACE_COUNT) {
            uint8_t index = getInterfaceIndex(command);
        if (!fifo16_isfull(&packet_starts[index]) && queued_bytes[index] < (getQueueSize(index))) {
            uint16_t s = current_packet_start[index];
            int16_t e = queue_cursor[index]-1; if (e == -1) e = (getQueueSize(index))-1;
            uint16_t l;

            if (s != e) {
                l = (s < e) ? e - s + 1: (getQueueSize(index)) - s + e + 1;
            } else {
                l = 1;
            }

            if (l >= MIN_L) {
                queue_height[index]++;

                fifo16_push(&packet_starts[index], s);
                fifo16_push(&packet_lengths[index], l);
                current_packet_start[index] = queue_cursor[index];
            }

        }
    }

  } else if (sbyte == FEND) {
    IN_FRAME = true;
    command = CMD_UNKNOWN;
    frame_len = 0;
  } else if (IN_FRAME && frame_len < MTU) {
    // Have a look at the command byte first
    if (frame_len == 0 && command == CMD_UNKNOWN) {
        command = sbyte;
        if  (command == CMD_SEL_INT0 
                 || command == CMD_SEL_INT1 
                 || command == CMD_SEL_INT2 
                 || command == CMD_SEL_INT3 
                 || command == CMD_SEL_INT4 
                 || command == CMD_SEL_INT5 
                 || command == CMD_SEL_INT6 
                 || command == CMD_SEL_INT7 
                 || command == CMD_SEL_INT8 
                 || command == CMD_SEL_INT9 
                 || command == CMD_SEL_INT10 
                 || command == CMD_SEL_INT11) {
            interface = getInterfaceIndex(command);
        }

    } else if  (command == CMD_INT0_DATA 
             || command == CMD_INT1_DATA 
             || command == CMD_INT2_DATA 
             || command == CMD_INT3_DATA 
....
```

like what is the syntax ? how do I read the config (frequency) ? how do I write the config  ? the FEND smells like FRAME END which mean PACKET 

anyway this is just a rant.I will probably end up figuring it out but if you know stuff that I don't, feel free to light the way.

in the meanwhile I will dig deeper

nathmo, 2024-08-11:
it seems to me that when the host set the frequency the RNode will send that to the radio. and when the host read the frequency it come from the eeprom ??? I must have missed something

jacobeva, 2024-08-13:
> it seems to me that when the host set the frequency the RNode will send that to the radio. and when the host read the frequency it come from the eeprom ??? I must have missed something

Correct, except when the host reads the frequency, it comes from the value stored in the object of the modem it is targeting.

To explain, a typical KISS packet would look something like `FEND 0x06 0x01 FEND`. Please see [here](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Framing.h) for more info on the framing.

However, I don't think the issue lies here. I would say there is an issue with your SPI communication. Are you sure the pins you have set in the config are correct? It is also worth mentioning the pin config does not take affect if `default_spi` is true, which in your case I think it is currently. Please try setting that to False, if you haven't already.

jacobeva, 2024-08-13:
The KISS framing is certainly something I will be documenting in the future. As you say, it is rather confusing to those not in the know.

nathmo, 2024-08-14:
and to be sure, the kiss standard is used to send the packet between the computer and the RNode even when the normal mode is used ?

jacobeva, 2024-08-14:
Not sure what you mean by normal mode. Do you mean connected to RNS through a serial port for example? If so, yes.

nathmo, 2024-08-18:
Exactly. (by normal mode I meant even when it's not in TNC mode, since I assumed that the KISS standard was related to TNC)

jacobeva, 2024-08-21:
Hi, did you manage to get this working @nathmo ?

nathmo, 2024-08-21:
No but for lack of trying. I'm in a motivation pitt... might try again later

jacobeva, 2024-10-22:
Hello again, have you had time to work on this at all? :)

nathmo, 2024-10-23:
No :( i'm back at school and had no time to fiddle around. I will likely give it another crack this summer

[1 more reply]

closed 2025-07-25 by jacobeva