liberatedsystems/RNode_Firmware_CE #19: LILIGO T-ECHO support

dgudim, 2024-08-03:
I want to install RNode on a T-Echo. Looking at the readme looks like its tranciever is supported (SX1262). But I am confused about the 
`These also must have an SPI interface and DIO_0 pin connected to the MCU directly.`
From the schematic, it looks like the SPI is connected directly, but I don't see any DIO_0 pins.
Here is the schematic: https://raw.githubusercontent.com/Xinyuan-LilyGO/T-Echo/main/T-Echo_Schematic.pdf

jacobeva, 2024-08-03:
Any of the DIO pins can actually be used, so this isn't a problem (but may require some tweaking in the FW). Please note that sometimes DIO_0 is referred to as DIO_1, as in the [SX1262 datasheet](https://www.rfsolutions.co.uk/downloads/1537522406DS_SX1261-2_V1.1_SEMTECH.pdf) for example, this means that you likely wouldn't have to change anything in the FW to get this to work, only write a board config. A classic case of zero-based indexing being used in some places and not others.

dgudim, 2024-08-03:
Cool! I'll try to write a board config then

jacobeva, 2024-08-03:
In case you have missed it, check CONTRIBUTING.md for instructions in the Documentation folder!

jacobeva, 2024-08-30:
Hello, have you managed to get anywhere with this? :)

jeremybox, 2024-09-09:
https://github.com/liberatedsystems/RNode_Firmware_CE/pull/35 submitted

dgudim, 2024-09-09:
Sorry for not replying earlier, I couldn't find the will/time to work on this (work + university = 💀)
@jeremybox I see you already implemented support, can I help you in any way?

jeremybox, 2024-09-09:
@dgudim Life happens, no worries.  I intend to work on the display stuff next.  If you want to go ahead and start looking at the bluetooth that would be great, but I'll also get to that eventually.

jacobeva, 2024-09-10:
Just to chime in, I assume that device only has BLE. If so, I have a feeling my existing NRF52 BLE implementation may work on it with a little tweaking perhaps.

jeremybox, 2024-09-12:
https://github.com/liberatedsystems/RNode_Firmware_CE/pull/36 replaces https://github.com/liberatedsystems/RNode_Firmware_CE/pull/35

jacobeva, 2024-09-12:
#36 has been merged, but we are still missing support for the display, firmware hash calculation and BLE.

jacobeva, 2024-09-12:
This compiler warning needs to be investigated too, this may affect CSMA:
`#warning "no hardware random number source detected for this platform"`

jacobeva, 2024-10-22:
Support for the T-ECHO has been merged, closing

closed 2024-10-22 by jacobeva