liberatedsystems/RNode_Firmware_CE #27: Is the 2.4 icon supposed to be blank?

FountainOfInk, 2024-09-01:
Hello, I am using a LoRa32 V2.1_1.6 (no TCXO, sx1276 variant (868/915/923MHz)). I have noticed that the 2.4ghz icon **is not** present. However, with markqvist's firmware (i.e. the firmware installed via `rnodeconf --autoinstall`), this icon **is** present. Is this behavior intentional? If not, should I open a bug report? I am more than willing to look at the code to fix this myself. 

### P.S. (Somewhat related)

Is there a conveinent way to convert the graphics in Graphics.h to bmp/png/whatever files? I am aware of https://javl.github.io/image2cpp/, but this requires knowing the size (the x and y dimensions) of each graphic in advance, which is not immediately obvious. It is possible to look at the `drawBitmap` calls in Display.h, but often it is not entirely possible to directly deduce the dimensions from that alone (or maybe I'm missing something obvious.)

In this image, the community firmware is on top, and markqvist's on the bottom.
[image]

jacobeva, 2024-09-01:
Yes, the lack of the 2.4GHz icon is intentional. To give some backstory, basically ages ago Mark intended on adding the ability to use the Wi-Fi chip on the ESP32 boards (like the one you have) as an interface, much like the LoRa modem on the board. However, this was never actually finished. When I performed the overhaul to add support for multiple interfaces to this firmware, I removed that icon as it was misleading, given that there was actually no code in place to allow for using the 2.4GHz modem. You're not missing out on a feature, don't worry ;)

With regards to your second request, I mainly just used image2cpp when doing work on the graphics (scaling them for the EINK display for the RAK platform). I don't have the original graphics, but I know who does. @markqvist, would you kindly upload all those graphics you produced somewhere so I can include them in this repository?

closed 2024-09-01 by jacobeva

FountainOfInk, 2024-09-02:
> When I performed the overhaul to add support for multiple interfaces (currently only multiple LoRa modems) to this firmware, I removed that icon as it was misleading, given that there was actually no code in place to allow for using the 2.4GHz modem. You're not missing out on a feature, don't worry ;)

Ah, I see. Thank you for clarifying. Out of curiosity, where might one start if they wanted to start working on an esp32 2.4ghz modem implementation? This is a feature that seems very interesting and useful to me, but I am not very well acquianted with the layout of the RNode firmware source code.

jacobeva, 2024-09-02:
Okay, firstly it's not going to be easy, so prepare yourself lol, but I will be here if you need me. If you need to contact me directly to ask about the implementation, you can always do so over Matrix (or if you want my LXMF address, just PM me on Matrix).

To begin, you'll need to research ad-hoc Wi-Fi, which is what will need to be used by the Wi-Fi chip in order to route Reticulum packets. I don't even know if the Wi-Fi chipset in your board supports it (but it probably does I would say). I'd recommend you try creating an example arduino sketch where you mess around with it and try sending data from one RNode to another using it. Then, you can go ahead and create a class in `Radio.cpp` to actually begin integrating it into the firmware, and test it after :)

That's a general idea of what I think you'll have to do. I think it would probably also be possible to connect it to Wi-Fi networks and communicate using IPv6, but I would try the ad-hoc stuff first. It will be easier to implement. Plus I know nothing about IPv6 so I wouldn't be much help haha.

If you think you're up to the task, I can reopen this issue and assign it to you (for implementing a Wi-Fi ad-hoc driver).

FountainOfInk, 2024-09-02:
> Okay, firstly it's not going to be easy, so prepare yourself lol
Ahaha; I expected as much. I would like to take a look, but I don't entirely expect to succeed. After a brief look, it seems like the esp32 doesn't support ad-hoc wifi, but the ESP-MESH project looks interesting (though it seems to be an uncomfortably large dependency to introduce into a mostly self contained project.) I'll contact you on Matrix! :)

jacobeva, 2024-09-02:
Hmm, maybe it is possible to do something here though? I wonder how ESP-MESH does it...

jacobeva, 2024-09-02:
[This is most likely how we can implement it](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html).

markqvist, 2024-09-02:
Please hold your horses on that one for a bit. I've been working on it for quite some time. The short version is that the obvious ways of going about it would fall so far short of the potential there is here, given the extra bandwidth available on raw 802.11 frames.

The way I am currently implementing it will really be quite a step forward, I think. I won't go into all the details here, but think of a fast and efficient, self-organising packet switching layer designed and optimised for Reticulum. Sort of like Ethernet and MPLS had a cryptographic little child together.

markqvist, 2024-09-02:
Or, I mean, if anyone wants to go ahead and add it in basic RNode mode, but just over raw 802.11, just go ahead of course! Just wanted to give a fair heads-up that I do have something in the pipeline in that regard :)

jacobeva, 2024-09-03:
I think we will leave this one in Mark's hands for now, and see where we end up. No use writing something which will be replaced soon anyway :)

FountainOfInk, 2024-09-03:
Sounds good. :)