liberatedsystems/RNode_Firmware_CE #32: Intermittent SX1280 packet errors

jacobeva, 2024-09-04:
Despite packets having to pass CRC, somehow packets are sometimes received which are completely invalid. For an example, see below:
```[2024-09-04 17:29:31] [Debug] Received invalid announce for <aa94e097d2d22b2b23343f746104f01c>: Invalid signature.
[2024-09-04 17:29:31] [Debug] Received invalid announce for <aa94e097d2d22b2b23343f746104f01c>: Invalid signature.```

This happens intermittently. It should be noted that this was a double announce on two RNode interfaces at the same time. It may be this is some kind of race condition or something. I've caught the raw packet from one of these before and literally 1 or 2 bytes will be incorrect, making me think it could also be some kind of memory issue.

Currently I have no clue what causes it, but it happens rarely enough for it to not be a problem most of the time.

jacobeva, 2024-09-10:
1 week later... I basically figured it out.

A large issue with the SX1280 is the sheer number of interrupts that it generates when operating at 203kbps. Currently I can't think of a great solution for this as I can't just sit polling the modem for ages since the firmware needs to be doing other stuff. For now, I'll leave this issue open until I think I can tackle it properly.

jacobeva, 2024-09-11:
Commit 51afd31 has temporarily addressed this by capping the data rate. The data rate can be increased again once this issue has been resolved, but for now this allows for greater reliability.

jacobeva, 2025-01-30:
Work is ongoing in the upstream to increase RNode throughput, it looks like the packet errors are currently caused by timing issues due to the high speeds. Closing for now as @unsignedmark is actively working on it.

closed 2025-01-30 by jacobeva

markqvist, 2025-01-30:
Yes, and to anyone else reading this, redesigned CSMA handling released in 1.81 upstream lays the groundwork for achieving the full throughput that SX1280 modems are capable of, but the current timings need to be optimised specifically for the much shorter frame times possible on those modems. To do that, I will need to get a more capable SDR and/or a more capable oscilloscope, but it that should hopefully happen down the line :)

In the meantime, the increased available bandwidth on fast SX1280 settings will still allow for more clients to coexist on the same shared channel, so all in all, it's still a win, even though we don't have the full speeds available for single clients yet.