liberatedsystems/RNode_Firmware_CE #33: python3 issue with makefile

faragher, 2024-09-07:
When doing make -upload-X on Debian 12 or similar systems that lack "python" redirecting to "python3" breaks when esptool.py is called. I don't have a good solution for this as while symlinking python -> python3 was long considered good practice, apparently at least Debian is moving away from it.

tomuk5, 2024-09-08:
Having this same issue, installing the package `python-is-python3` creates this symlink on debian systems without manual intervention.

```sudo apt-get install python-is-python3```

In the Long-term all references to `python` should be replaced with `python3` then this transitional package can be removed.

jeremybox, 2024-09-09:
One of the changes in https://github.com/liberatedsystems/RNode_Firmware_CE/pull/35 addresses this.

jacobeva, 2024-09-10:
That's pretty gay.

> In the Long-term all references to python should be replaced with python3 then this transitional package can be removed.

Agreed.

tomuk5, 2024-09-11:
[PEP394](https://github.com/hroncok/peps/blob/master/pep-0394.txt) covers this eventuality:

In order to tolerate differences across platforms, all new code that needs
to invoke the Python interpreter should not specify ``python``, but rather
should specify either ``python2`` or ``python3`` (or the more specific
``python2.x`` and ``python3.x`` versions; see the `Migration Notes`_).
This distinction should be made in shebangs, when invoking from a shell
script, when invoking via the system() call, or when invoking in any other
context.

jacobeva, 2024-09-11:
Thanks for the info!

commit f8299568 (2024-09-11): Replace all references to python with python3 instead, close #33
closed 2024-09-11 by jacobeva