liberatedsystems/RNode_Firmware_CE #48: I'm getting compiling error

eUGENE-BNX, 2024-10-10:
Hi there,

I am trying to compile with the Arduino IDE and I am getting the following error.
Im using "http://unsigned.io/arduino/package_unsignedio_UnsignedBoards_index.json"

how can I fix that?

In file included from /Users/eugene/Documents/Arduino/RNode_Firmware_CE/Radio.hpp:13:0,
                 from /Users/eugene/Documents/Arduino/RNode_Firmware_CE/Utilities.h:16,
                 from /Users/eugene/Documents/Arduino/RNode_Firmware_CE/RNode_Firmware_CE.ino:18:
/Users/eugene/Documents/Arduino/RNode_Firmware_CE/Boards.h:60:8: error: #error "The firmware cannot be compiled for the selected MCU variant"
       #error "The firmware cannot be compiled for the selected MCU variant"
        ^~~~~
In file included from /Users/eugene/Documents/Arduino/RNode_Firmware_CE/RNode_Firmware_CE.ino:18:0:
/Users/eugene/Documents/Arduino/RNode_Firmware_CE/Utilities.h:20:10: fatal error: algorithm: No such file or directory
 #include <algorithm>
          ^~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: #error "The firmware cannot be compiled for the selected MCU variant"

jacobeva, 2024-10-10:
Hi Alp, can you show me the make command you're using please?

eUGENE-BNX, 2024-10-10:
Hi jacobeva,

I just tried to verify/compile via Arduino IDE, should I compile via console?

jacobeva, 2024-10-10:
Yes. Please see the files in the documentation folder, namely BUILDING.md

eUGENE-BNX, 2024-10-10:
sorry , read the f'ckn document first :)

Ok, compiled...  Thanks.. I will try to upload now ...

eUGENE-BNX, 2024-10-10:
as i expected ,Upload Error :)

make upload-lora32_v21
arduino-cli upload -p /dev/ttyACM0 --fqbn esp32:esp32:ttgo-lora32

makes an error like as blow;
 [Errno 2] No such file or directory: '/private/var/folders/7z/4c6r524d66s6lh4mhdbm06fr0000gn/T/arduino/sketches/28578C8ACC75FD4CB50A7321C52A58F6/RNode_Firmware_CE.ino.bootloader.bin'

jacobeva, 2024-10-10:
Are you using macOS for flashing? I don't recognise that directory structure.

eUGENE-BNX, 2024-10-10:
Yes unfortunately Im on macOS now

jacobeva, 2024-10-10:
I see. I'm not sure why it's trying to flash what seems to be the bootloader on that board, but I don't personally have one so that might be normal :)

What is definitely not normal is arduino's inability to find the file it's looking for. The only thing I can suggest is trying to reinstall arduino-cli on your computer, or use another computer running a different OS (I recommend Linux generally). Unfortunately this seems like an issue in arduino-cli itself, which wouldn't be caused by anything I've set in the makefile.

jacobeva, 2024-10-10:
Just to be 100% sure, did you run `make firmware-lora32_v21` before trying to upload the firmware?

eUGENE-BNX, 2024-10-10:
Thank you for all your help.. I will also try with Ubuntu OS tonight and inform you.. 

btw; macos uses "/dev/cu.usbserial-56570103781" instead of "/dev/ttyACM0" in naming.. its wierd :)

jacobeva, 2024-10-10:
So I've heard...

jacobeva, 2024-10-10:
> Whats wrong with me?

I ask myself that question rather often...

Joking aside, have you replaced them in [this section](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Graphics.h#L16)? Don't replace them in the other section, that's only for 122px wide displays, which I'm sure you don't have :)

I know you've probably done this anyway, but just to check, can you please ensure you run `make firmware-[target]` each time you make changes, and then upload the firmware? It's possible you're doing this anyway, but just to be sure ;)

eUGENE-BNX, 2024-10-10:
Im working on it :)

should the graphics be 122x64? is there a way to understand which area should be how many pixels?

jacobeva, 2024-10-10:
The 122px wide graphics are currently only for the RAK4631 board. It
seems like you're more looking for help at this point, do you mind if I
email you the link to the Reticulum matrix room? It would be easier and
more appropriate to help you there.

jacobeva, 2024-10-10:
Providing assistance over Matrix. Not an issue in the FW so closing...

closed 2024-10-10 by jacobeva