liberatedsystems/RNode_Firmware_CE #53: T-echo reports TX power 0 dBm

fustygerbil, 2024-12-18:
When trying to start nomadnet, T-echo (CE firmware 1.73) reports TX power of 0 dBm

`[2024-12-18 12:45:38] [Debug] Started shared instance interface: Shared Instance[37428]
[2024-12-18 12:45:38] [Debug] Cleaning ratchets...
[2024-12-18 12:45:38] [Verbose] Bringing up system interfaces...
[2024-12-18 12:45:38] [Notice] Opening serial port /dev/ttyACM0...
[2024-12-18 12:45:41] [Notice] Serial port /dev/ttyACM0 is now open
[2024-12-18 12:45:41] [Verbose] Configuring RNode interface...
[2024-12-18 12:45:41] [Verbose] Waiting for radio configuration validation for RNodeInterface[RNode LoRa Interface]...
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting frequency is 867.2 MHz
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting bandwidth is 125.0 KHz
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting symbol time is 2.05ms (at 488 baud)
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting preamble is 22 symbols (45ms)
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting CSMA slot time is 50ms
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting TX power is 0 dBm
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting spreading factor is 8
[2024-12-18 12:45:41] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting coding rate is 5
[2024-12-18 12:45:41] [Verbose] RNodeInterface[RNode LoRa Interface] On-air bitrate is now 3.12 kbps
[2024-12-18 12:45:41] [Notice] RNodeInterface[RNode LoRa Interface] is configured and powered up
[2024-12-18 12:45:41] [Verbose] System interfaces are ready
[2024-12-18 12:45:41] [Verbose] Utilising cryptography backend "openssl, PyCA 44.0.0"
[2024-12-18 12:45:41] [Verbose] Configuration loaded from /home/forest/.reticulum/config
[2024-12-18 12:45:41] [Verbose] Loaded 1 known destination from storage
[2024-12-18 12:45:41] [Verbose] Loaded Transport Identity from storage`

jacobeva, 2024-12-18:
Can you paste your interface config file here please? It looks like this configuration of 0dBm is actually set in your config file. If it wasn't, I would expect RNS to complain that the TX power does not match the configured value.

fustygerbil, 2024-12-19:
Sorry, I should have said that I saw the TX power mismatch error with any (config file) txpower value > 0. This situation applies regardless of the configured (non-negative integer) value (even though I did have it set to 0, here, to see what would happen).

Incidentally, running rnodeconf --info returns a maximum TX power level of 22 dBm.

jacobeva, 2024-12-23:
@jeremybox any ideas mate?

jacobeva, 2025-02-03:
@fustygerbil Can you confirm if you still have this issue on the current `dev` branch?

jeremybox, 2025-02-05:
I have always used 0db in my device, so I'm not sure what it "should" be.

https://github.com/markqvist/Reticulum/blob/fd6d8ffff82d75f52fe3f3f9c5726a24f55ee4f0/RNS/Utilities/rnodeconf.py#L279

The above is the code that rnodeconf --info returns.  It's just a hardcoded array of numbers.  When I added the T-Echo code I just copied that 22 from the other lines because I didn't know what that value meant, but the 22dBm isn't polling the actual device itself.  Will have to check out the chip's actual max power from spec when I get some time, but I have the same result in a quick test where anything besides 0dBm doesn't function.

jacobeva, 2025-02-12:
That's odd.

fustygerbil, 2025-02-25:
> [@fustygerbil](https://github.com/fustygerbil) Can you confirm if you still have this issue on the current `dev` branch?

Sorry, I'm failing to get working firmware on the device now. After DFU upload, `rnodeconf /dev/ttyACM0 --firmware-hash $(./partition_hashes from_device /dev/ttyACM0)` fails because `partition_hashes from_device /dev/ttyACM0` returns `Could not get partition hash from device: list index out of range`

serialrf433, 2025-12-25:
Does the issue still exist with the latest RNode release from here? https://github.com/markqvist/RNode_Firmware/releases/tag/1.85

jacobeva, 2026-08-20:
Anyone with a T-Echo know if this is still reproducible?

jeremybox, 2026-08-25:
I can look into this next week when I get back from a work trip.  Adding a calendar reminder for myself.