liberatedsystems/RNode_Firmware_CE #56: Looking for a guide/manual on how to edit/compile/flash for newbies

tokio69, 2025-01-14:
Hello, 
I want to build a Rnode PCB from scratch. I am not used to github and compiling stuff and I see there is no "flash-ready" firmware for my hardware build. 
Where can i find a step-by-step manual on how to adjust what file (like pin numbers for example) starting from a blank ubuntu computer to a working and configurable Rnode? In the manual it states "compile it" but nowhere on HOW to compile/download/edit/ and flash something.
The build is based on ESP32 + Ebyte E22-400M33S tranceiver with SX12xx chip. I am HAM licenced and want to introduce this to all the other hammers in my location. Most of them tell me the same thing, they don't want to start with it cause it is too complicated for them to get started.  The documentation now is writen for a skilled programmer/linux/git specialists who know how to interpret some loose CLI commands, otherwise you are starting a very long learning curve with trial and error to get a results.
Hope some builders here have better programming skills then mine and have something writen out or Liberated systems can enlarge their README with more detailed information.

jacobeva, 2025-01-26:
Hi Tom,

Here's a [link to the contributing page.](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Documentation/CONTRIBUTING.md) I suggest you read it, it will likely be helpful to you. I am of course assuming you have not done this already. Regarding building (compiling), see [here](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Documentation/BUILDING.md).

Regarding things being difficult to use, yes I agree. If you could give any feedback on the current documentation (be specific, is detail lacking somewhere, are some things inaccurate, etc) I will be happy to make improvements. However, there *is* a certain level of knowledge required for someone to be able to use Git, edit C header files, etc to be able to contribute to this repository, or modify it to write their own board config. Once such support for a board is added, it is possible for future users to install the firmware on a similar board by using a tool such as `rnodeconf`. Yes, it is command line, but in my opinion it is usually quite straightforward to use. There's also Liam Cottle's [web flasher](https://liamcottle.github.io/rnode-flasher/).

I am working on making flashing RNodes easier through some of my current work which is not public yet.

jacobeva, 2025-02-14:
Closing this as it's been 3 weeks with no reply.

closed 2025-02-14 by jacobeva