liberatedsystems/RNode_Firmware_CE #58: Outline board definition for Seeed Xiao S3 WIO board.

easytarget, 2025-02-13:
Last month I had a (failed) attempt to add the Seeed studio Xiao S3 module to `Boards.h`, I did some work, but hung up on building it and was forced to move on to other stuff. Since I did that there has been some light re-factoring on boards.h and I'm not even sure this is the correct format for pin definitions anymore.

However; I dont want to simply ditch the research I did. So I'm dumping it here in case anybody can do this properly. I'd be able to test this if it can be built. The pin definitions have been checked against those used in meshtastic and loraWan clients for this module and match.

+    https://files.seeedstudio.com/products/SenseCAP/Wio_SX1262/Wio-SX1262_Module_Datasheet.pdf
+    https://files.seeedstudio.com/products/SenseCAP/Wio_SX1262/Schematic_Diagram_Wio-SX1262_for_XIAO.pdf
+    https://files.seeedstudio.com/wiki/SeeedStudio-XIAO-ESP32S3/res/XIAO_ESP32S3_SCH_v1.2.pdf
+    Module uses dio2 for rx enable but also has external RXEN on IO38
+    There is a user LED on IO48 (1==on)
+    There is a user button on IO21 (needs pullup)

I got as far as a definition in `Boards.h`
 ```c
   #elif BOARD_MODEL == BOARD_SEED_XIAO_S3_WIO
      #define IS_ESP32S3 true
      #define MODEM SX1262
      #define DIO2_AS_RF_SWITCH true
      #define HAS_BUSY true
      #define HAS_TCXO true

      #define HAS_DISPLAY false
      #define HAS_CONSOLE true
      #define HAS_BLUETOOTH false
      #define HAS_BLE true
      #define HAS_PMU false
      #define HAS_NP false
      #define HAS_SD false
      #define HAS_EEPROM true

      #define HAS_INPUT false
      #define HAS_SLEEP false

      #define I2C_SCL 6
      #define I2C_SDA 5

      const int pin_btn_usr1 = 21;

      const int pin_cs = 41;
      const int pin_reset = 42;
      const int pin_sclk = 7;
      const int pin_mosi = 9;
      const int pin_miso = 8;
      const int pin_tcxo_enable = -1;
      const int pin_dio = 39;
      const int pin_busy = 40;

      const int pin_led_rx = 48;
      const int pin_led_tx = 48;
```

and an entry in sx126x.cpp for TCXO voltage:
```c
     #elif BOARD_MODEL == BOARD_TECHO
       uint8_t buf[4] = {MODE_TCXO_1_8V_6X, 0x00, 0x00, 0xFF};
+    #elif BOARD_MODEL == BOARD_SEEED_XIAO_S3_WIO
+      uint8_t buf[4] = {MODE_TCXO_1_8V_6X, 0x00, 0x00, 0xFF};

```

jacobeva, 2025-02-14:
Nice, thank you. I think I own this board myself so this is helpful information.

cobraPA, 2025-02-24:
There is a PR for this in Mark's upstream repo: https://github.com/markqvist/RNode_Firmware/pull/106

jacobeva, 2025-03-17:
@cobraPA Thank you, I will incorporate that into this repo once I have it working.

jacobeva, 2025-03-19:
Closed by ea77f3b32e6eae68a9bfd2290d1c6efb62af071d

closed 2025-03-19 by jacobeva