liberatedsystems/RNode_Firmware_CE #60: Heltec V3 Battery indicator not working

stickyvalentine, 2025-02-26:
I have a pair of Heltec V3 boards that came bundled with MakerFocus branded 3000mAh Lithium batteries and the battery indicator is always empty regardless of how much charge they have. I'm too embarrassed to say where I bought them but I will say they came with prime shipping 😰. I just flashed them a few days ago with a fresh RNS install using rnodeconf on Linux 

If I can provide any additional information or help in any way let me know. This isn't a great picture, but I hope it gets the idea across.

[image]

gofaster, 2025-03-19:
If they have the Heltec Lora32 V3.2 revision; it seems that the voltage detection scheme has changed.  I have a couple of v3.2 and battery level is not displayed either.
See [http://community.heltec.cn/t/heltec-v3-1-vs-v3-2-what-changed/17407/4](http://community.heltec.cn/t/heltec-v3-1-vs-v3-2-what-changed/17407/4) and the next post.

jacobeva, 2025-03-27:
Thank you for the information, I will look into this soon.

cobraPA, 2025-05-10:
Curious about the case and battery you mentioned.  What country did you order these in?

jacobeva, 2026-08-20:
This should be in the next release.