liberatedsystems/RNode_Firmware_CE #61: Add SX1268 support (E22-400M33S) module

pa2eon, 2025-02-27:
Dears,

Is it possible to handle the next HW combination;

ESP: espressif-esp32-s3-n16r8
Lora module: E22-400M33S - with SX1268 RX/TX board.

We have tested this combination with the next connected lines:

E22 pins  -  Name - ESP32 pins

13      -   DIO1    -     15
14      -   Busy     -     nil
15      -   NRST    -     14
16      -   MISO    -     13
17      -   MOSI    -     11
18     -    SCK       -     12
19     -    NSS      -      10  

But after compiling the ESP dit not see the firmware ..
So far I can see is in de source code no entry for an SX1268 chip.

Is it an idea to make a short steps document .. what code I have to add to which file .. to bring this combination to live.

Regards,  Eric

victorfmin66, 2025-03-04:
Hello Eric, the chip E22-400M33S is not using the SX126x Semtech chip, is using the LLCC68 internally. That is the reason why is not compatible at this time. The LLCC68 chip is very similar as SX126x but only focused on LoRa modulation variables while SX126x accept other modulations such as FSK. I don't think is very difficult to implement the LLCC68 on this project as in RadioLib was very easy to implement.

Sincerely.

Victor

tokio69, 2025-03-05:
Is this the same with the 400M30S or 900M30S module @victorfmin66 ?

joshbowyer, 2025-04-10:
> Is this the same with the 400M30S or 900M30S module [@victorfmin66](https://github.com/victorfmin66) ?
> 
> [image]

The E22-900M30S and 22S are SX1262 radios with PA/LNA embedded.

pa2eon, 2025-04-10:
Dears,

Go to https://wiki.hamesh.eu and then links.
Check the 'meezenest ... '  link and you read our development

Met vriendelijke groeten,

Regards,

Eric Oosterbaan - PA2EON

 ---- Op do, 10 apr. 2025 16:32:49 +0200  schreef ***@***.***  ----

joshbowyer left a comment https://github.com/liberatedsystems/RNode_Firmware_CE/issues/61#issuecomment-2793815333

Is this the same with the 400M30S or 900M30S module https://github.com/victorfmin66 ?

https://github.com/user-attachments/assets/48d49b16-0820-4845-ab5d-26bfea00735f

The E22-900M30S and 22S are SX1262 radios with PA/LNA embedded.

jacobeva, 2025-04-10:
Dank je wel Eric, but can you please provide a direct link? I cannot find what you speak of on your site, and my Dutch is too poor to help me in this regard ;)