liberatedsystems/RNode_Firmware_CE #64: Create supported boards list

jacobeva, 2025-03-09:
A supported boards list with any information on certain incompatibilities or long standing issues with boards should be created for this repo. It should ideally be in markdown.

brybalicious, 2026-04-18:
Yes. It would be great to build up to a web flasher similar to Meshtastic or Meshcore so that onboarding and scaling Reticulum use is that much easier... difficult now that the main repos are closed to PR's...

https://flasher.meshtastic.org/
https://flasher.meshcore.io/

gh4chris, 2026-10-06:
For most of the supported device types, it is also possible to use
[Liam Cottle's Web-based RNode Flasher](https://liamcottle.github.io/rnode-flasher/).
This option may be easier if you're not familiar with using a command line interface.

The RNode Firmware supports the following boards:

- LilyGO T-Beam v1.1 devices with SX1276/8 LoRa chips
- LilyGO T-Beam v1.1 devices with SX1262/8 LoRa chips
- LilyGO T-Beam Supreme devices
- LilyGO T-Deck devices (currently display is disabled)
- LilyGO LoRa32 v1.0 devices
- LilyGO LoRa32 v2.0 devices
- LilyGO LoRa32 v2.1 devices (with and without TCXO)
- LilyGO T3S3 devices with SX1276/8 LoRa chips
- LilyGO T3S3 devices with SX1262/8 LoRa chips
- LilyGO T3S3 devices with SX1280 LoRa chips
- LilyGO T-Echo devices
- Heltec LoRa32 v2 devices
- Heltec LoRa32 v3 devices
- Heltec LoRa32 v4 devices
- Heltec T114 devices
- RAK4631 devices
- SeeedStudio XIAO ESP32S3 devices (with Wio-SX1262)
- Homebrew RNodes based on ATmega1284p boards
- Homebrew RNodes based on ATmega2560 boards
- Homebrew RNodes based on Adafruit Feather ESP32 boards
- Homebrew RNodes based on generic ESP32 boards

## Supported Transceiver Modules

The RNode Firmware supports all transceiver modules based on Semtech **SX1276**, **SX1278**, **SX1262**, **SX1268** and **SX1280** chips, that have an SPI interface and expose the relevant DIO interrupt pins from the chip.