liberatedsystems/RNode_Firmware_CE #65: Wrong RX/TX switching for BOARD_E22_ESP32 variant?

azidoazid, 2025-03-14:
I am currently trying to compile my own rnode firmware for the following hardware:
https://github.com/Hydra-Designs/project-hydra-meshtastic-pcb
And while browsing

azidoazid, 2025-03-14:
I am going to give it a try, should be straight-forward to adapt the RX/TX switching from the sx128x driver. I'll report back in a couple of days, I really appreciate the building.md documentation!

azidoazid, 2025-03-17:
I had some troubles with arduino-cli and the library management https://github.com/liberatedsystems/RNode_Firmware_CE/issues/66. Somehow I ended up with the most recent libraries, even though I followed the guide and did a `make prep-esp32` beforehand. Since libs are installed system-wide I must have managed to update the libs in between downloading+prepping and then compiling.

My modified firmware is building with no errors, however `make upload-e22_esp32` fails after uploading with:
`FileNotFoundError: [Errno 2] No such file or directory: './build/esp32.esp32.esp32/RNode_Firmware_CE.ino.bin`

I manually adjusted the output dir:
https://github.com/azidoazid/RNode_Firmware_CE/blob/db046ec53163be2f8d37cac4b37416b8a6e01e25/Makefile#L99

And adjusted the path in the make upload step:
https://github.com/azidoazid/RNode_Firmware_CE/blob/db046ec53163be2f8d37cac4b37416b8a6e01e25/Makefile#L257

Now I get the following error message:

> Opening serial port /dev/ttyUSB0...
> Device connected
> Current firmware version: 1.74
> Reading EEPROM...
> This device has not been provisioned yet, cannot set firmware hash
> make: *** [Makefile:257: upload-e22_esp32] Error 77

And the OLED shows "MISSING CONFIG!"

I also tried running:
`rnodeconf /dev/ttyUSB0 -r --platform ESP32 --model a6 --product f0 --hwrev 1
`
which outputs: 

> Opening serial port /dev/ttyUSB0...
> Device connected
> Current firmware version: 1.74
> Reading EEPROM...
> Waiting for ESP32 reset...
> Loading signing key...
> Bootstrapping device EEPROM...
> EEPROM written! Validating...
> Waiting for ESP32 reset...
> EEPROM checksum correct
> Device signature validated
> EEPROM Bootstrapping successful!

If I then manually re-run
`rnodeconf /dev/ttyUSB0 --firmware-hash $(./partition_hashes ./build/RNode_Firmware_CE.ino.bin)
`

> Opening serial port /dev/ttyUSB0...
> Device connected
> Current firmware version: 1.74
> Reading EEPROM...
> EEPROM checksum correct
> Device signature validated
> Firmware hash set

I then manually retried `make upload-e22_esp32` but I had to modify:  https://github.com/azidoazid/RNode_Firmware_CE/blob/db046ec53163be2f8d37cac4b37416b8a6e01e25/Makefile#L259 to use a recent version of esptool since the included esptool failed to enter bootloader mode on my hardware.
Afterwards `make upload-e22_esp32` finished with no errors but I am still left with "MISSING CONFIG". I made sure to run `esptool erase_flash` before trying all the above steps.

Any idea on what's going on?

azidoazid, 2025-03-26:
@jacobeva Any idea? I am still stuck at "MISSING CONFIG".

jacobeva, 2025-03-27:
This sounds like it doesn't like the EEPROM config you've given it, are you sure the product and model values are valid? I would check for you but that isn't possible for me right now.

azidoazid, 2025-03-31:
I've taken the product and model values from: https://github.com/liberatedsystems/RNode_Firmware_CE/blob/master/Documentation/BUILDING.md and tried several of those. I can also read the config with rnodeconf.
I also don't understand why `make upload` fails with ` This device has not been provisioned yet, cannot set firmware hash`, this clearly should not happen if the makefile is working as intended?

jacobeva, 2025-04-06:
> this clearly should not happen if the makefile is working as intended?

You are right to a certain extent. The makefile does need some refactoring to support board flashing when they haven't been provisioned yet. I've largely inherited the makefile from the upstream repo. PRs welcome.

jacobeva, 2025-04-06:
Use model `ff` and product `f0`. This should work now thanks to 40a97a4

tomuk5, 2025-04-29:
Hi, apologies for delay in replying on this.

I did deviate from the diagram slightly (and i need to update this) i use the DIO2 pin bridged to TXEN to save having to implement TXEN functionality for the SX1262.

I've just made a pull-req for revised settings for TCX0 and SPI pins that should result in a fully working E22 board (with the TXEN bridged to DIO2)