liberatedsystems/RNode_Firmware_CE #67: Bluetooth on iOS?

0x62, 2025-03-21:
Should Bluetooth on iOS be supported? I'm using the latest CE firmware (1.74). I've enabled BT and entered pairing mode, but nothing shows up on my iPhone 14. I can see and pair to the RNode on my Mac fine.

jacobeva, 2025-03-27:
To my knowledge cobraPA has created an RNode Firmware fork that works with iOS Bluetooth. I need to look into how he's done it.

https://github.com/cobraPA/RNode_Firmware_iOS

cobraPA, 2025-05-10:
Main difference is that the BLE framework in my version was picked for best compatibility with iOS BLE and ESP32 (and was done before any BLE support was in Mark's firmware.)  You will need to flash RNode firmware from my repo's for now if you are using Sideband for iOS.  @0x62, for Mac, do you know if the pairing was Classic Bluetooth or BLE?  Good news that works in some cases.  iOS has limitations on Classic Bluetooth use and we use the BLE interface there.

ronangaillard, 2025-08-24:
Can we expect IOS compatibility in the CE firmware soon ?

What should be done ?