liberatedsystems/RNode_Firmware_CE #68: Flash size for xiao esp32s3 is incorrect, board has 8Mb not 4Mb.

easytarget, 2025-03-24:
Commits ea77f3b and ea77f3b add support for the XIAO esp32s3 board with wio LoRa module. But assume the board has 4Mb of flash.

I noticed in the seedstudio docs that this board actually has 8MB flash on it. 

I don't know if this is as simple as just changing the value in the [Makefile](https://github.com/liberatedsystems/RNode_Firmware_CE/blob/7342116d91221391376cb7cc4e107d648c32a0d7/Makefile#L215) or if anything else would need adjusting in the partition table etc.
 
ref:
https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/
https://wiki.seeedstudio.com/wio_sx1262_xiao_esp32s3_for_single_channel_gateway/#flash-firmware

jacobeva, 2025-03-27:
Good catch!

Yes, it just needs changing in the makefile actually. It doesn't really affect anything as we don't use that much flash in the firmware.

A PR is welcome if you have time.

Thank you.

jacobeva, 2025-04-06:
#72 merged

closed 2025-04-06 by jacobeva