liberatedsystems/RNode_Firmware_CE #73: New board: Radiomaster 900 Bandit Nano

jayrahdevore, 2025-04-04:
I wish this were a PR, ready to go, but I don't own the hardware (yet).  I'm just wanting to paste information I've found for this board, either for anyone else looking to add this board or for myself at an unknown future date.

---

I recently came across the [Radiomaster 900 Bandit Nano board](https://www.radiomasterrc.com/products/bandit-nano-expresslrs-rf-module) and thought it would be an excellent candidate for RNode.  Originally designed for ExpressLRS R/C over LoRa, it:
- is commercially available at a reasonable price for a 1W unit (~$50)
- supports a wide input voltage range (6-16.8V)
- has active cooling
- uses the SX1276 transciever
- supports up to 1W output with some sort of PA
- already has an injection molded enclosure and an XT30 power connector
- has an ESP32 and supports bluetooth

---

With the stock ExpressLRS firmware, point-to-point connections should already be available via a [basic serial link](https://www.expresslrs.org/software/airport/)

Meshtastic appears to support it ([source code](https://github.com/meshtastic/firmware/blob/1b33189fe62d57b067b98facd84656531903457b/variants/radiomaster_900_bandit_nano/variant.h))([discussion](https://github.com/meshtastic/firmware/discussions/3794))

jacobeva, 2025-04-06:
Wicked! This is definitely a job for an American though. That is illegal to operate here in the UK haha.

cobraPA, 2025-05-10:
They are out of Hong Kong... interesting.