liberatedsystems/RNode_Firmware_CE #74: Version Numbering Parity?

LinuxinaBit, 2025-04-04:
Is version numbering parity with the upstream RNode_Firmware possible?
Possibly something like `X.XX-YY`?

jacobeva, 2025-04-06:
Already discussing this with Mark. My current proposal to him is to have a framework version and firmware version, which are separate. The framework version will be tied to the upstream rnode firmware version, whereas the firmware version will be a regular firmware version. This should fix issues with misunderstandings caused by firmware version disparity when using RNodes with RNodeInterface (when it checks the firmware version for compatibility).

jacobeva, 2025-04-07:
Please keep this open actually, it will be useful for tracking :)