liberatedsystems/RNode_Firmware_CE #75: Using Rnode over UART (RAK)

dswinton, 2025-04-05:
Hi Guys,

Does anyone know if it's possible to use an Rnode over UART?

I have a RAK4631 flashed with the latest community firmware.  
It's working fine over USB so the firmware and node itself are good.

My use case where im intending to install it doesn't have available USB, but I do have access to UART.

When I connect up via UART and try running...

```
rnodeconf --info /dev/ttyS4
[01:57:21 Opening serial port /dev/ttyS4...
[01:57:24 Serial port opened, but RNode did not respond. Is a valid firmware installed?
Got invalid response while detecting device
```

I've double checked the wiring and confirmed the UART board is good... so I'm assuming this means that UART on the RNode firmware isn't implemented?

Does anyone know if this is the case, or if there's a way to enable it?

Thanks in advance!

jacobeva, 2025-04-06:
Not implemented yet but definitely possible. It's probably called something like Serial1 or Serial2. Definitely one to look at.

jacobeva, 2025-04-06:
https://github.com/ReserveNetwork/T114-firmware/commit/4264a9c3679faa60e92ea8f96babb96c715a1476
https://github.com/ReserveNetwork/T114-firmware/commit/782a2e77e91f2ee4a9a78440fc68c529bc0e8351

This is an example of me implementing this for a different board in a different repo. Just in case someone wants to do a PR, this is how it can be done. Ideally, the extra serial port should be defined in Boards.h but I was rushing.