liberatedsystems/RNode_Firmware_CE #76: LR1110? (Seeed T1000-E)

joshbowyer, 2025-04-10:
This device has become very popular in the Meshtastic community and has very good tx/rx for its tiny antenna (as well as being small, light, and waterproof [IP65]). It uses an LR1110 radio chip. Is this a potential candidate for the Rnode firmware?

jacobeva, 2025-05-10:
I do have one of these. Currently the main issue is the driver (or lack thereof) and the ability to flash it since there currently isn't an arduino BSP for it. I believe @cobrapa was working on this and got quite far.

cobraPA, 2025-05-10:
I've posted binaries for Mark's RNode branches that you can flash now.  Firmware source is only available there so far.  As jacobeva says, there is not an upstream Arduino IDE target for the Seeed T1000-E.  It sounds like it may not being coming, as they produced it for the Meshtastic community, not as a generic open source product.  They have released an open-source variant as well recently, but it is targeted at using the Nordic dev tools directly...  Anyway, at some point I may put out a target for Arduino IDE, but it's not on the top of my list.  Binaries for Marks's version are here alone with flashing details:

- https://github.com/cobraPA/RNode_bin/tree/master/t1000e_mark

Source is available for that version as well (but as mentioned, you won't find a target in Arduino IDE to flash it.  I think there is a 'compatible' one available, but it is configured for a different USB ID, so will fail.)

- https://github.com/cobraPA/RNode_Firmware_iOS/tree/ble_nrf52_t1000e_cleanup

There is also an iOS BLE RNode version available.

jeremybox, 2025-07-31:
@cobraPA Care to share how you managed to build it?  I have one of these devices too and have never gotten anything to build.

cobraPA, 2025-07-31:
Not trivial to setup... If I had been smart I would have documented the process, but I had no idea what I was doing the first time so will have to recreate it to publish it.  Use the binary?

cobraPA, 2025-07-31:
Actually, this may be good news... It looks like Seeed may have made the Arduino bootloader available, so give that a shot... (I just saw it and have not tried it yet.)

https://wiki.seeedstudio.com/t1000_e_arduino_examples/

cobraPA, 2025-12-14:
I haven't seen anyone post builds other than the ones I gave above on May 10.  These should still be usable, I think.  They will get updated at some point.  My plan is to revisit that firmware with a microReticulum build in the future, but that may take a bit.

m0wer, 2025-12-14:
Thanks @cobraPA ! Very helpful.

I manage to connect to it from the app but it keeps saying `After configuring RNodeInterface[RNodeInterface], the reported radio parameters did not match your configuration.`. 

I've configured following the instructions from your repo and then with:

```
% python rnodeconf.py /dev/ttyACM4 --bluetooth-on --tnc --freq 869525000 --bw 250000 --sf 8 --cr 5 --txp 14
[17:40:36] Opening serial port /dev/ttyACM4...
[17:40:39] Device connected
[17:40:39] Current firmware version: 1.81
[17:40:39] Reading EEPROM...
[17:40:39] EEPROM checksum correct
[17:40:39] Device signature validated
[17:40:39] Enabling Bluetooth...

[17:40:40] Radio reporting frequency is 869.525 MHz
[17:40:40] Radio reporting bandwidth is 250.0 KHz
[17:40:40] Radio reporting TX power is 14 dBm
[17:40:40] Radio reporting spreading factor is 8
[17:40:40] Radio reporting coding rate is 5
[17:40:41] Device set to TNC operating mode
```

Also tried with normal mode instead of TNC. And set those same radio settings in the app.

Any idea of what could be failing?

m0wer, 2025-12-16:
Tried flashing the zip with nrfutil instead and to not set the frequency from the computer, just directly from sidenode Android app. And using 915 MHz, but still same error.

But it does work from the computer through serial!

```
  [[RNode LoRa T1000-E]]
    type = RNodeInterface
    enabled = yes
    port = /dev/ttyACM0
    frequency = 869525000
    bandwidth = 250000
    txpower = 14
    spreadingfactor = 8
    codingrate = 5
```

Juankar-Carrillo, 2025-12-17:
> Intenté flashear el archivo zip con nrfutil y configurar la frecuencia directamente desde la aplicación de Android de Sidenode, sin usar la computadora. Usaba 915 MHz, pero seguía con el mismo error.
> 
> ¡Pero funciona desde la computadora a través del puerto serial!
> 
> ```
>   [[RNode LoRa T1000-E]]
>     type = RNodeInterface
>     enabled = yes
>     port = /dev/ttyACM0
>     frequency = 869525000
>     bandwidth = 250000
>     txpower = 14
>     spreadingfactor = 8
>     codingrate = 5
> ```

As I mentioned, it works perfectly fine from the computer; I used it a few months ago for testing (also on the 868 band).

cobraPA, 2026-02-17:
> [@cobraPA](https://github.com/cobraPA) Hey cobra, are you interested in making a PR for this unit here?

Only mildly...  The difference in driver setup is significant and the LR11xx command format is multi-byte now, so the new RNodeCE architecture isn't something I've spent time seeing what it would take to make that work.  I'm also tied up with 3 other (or more) pieces of Reticulum related software, so I can't promise much time on an RNodeCE port for the near future.