liberatedsystems/RNode_Firmware_CE #88: Bug: LilyGO T-Echo unresponsive after flash (v1.75)

ronangaillard, 2025-08-24:
**Describe the Bug**

Yes, this is a reproducible bug affecting the LilyGO T-Echo. After a successful firmware flash of both official RNode firmware (v1.81+) and the Community Edition firmware (v1.75), the device becomes unresponsive to configuration tools. The primary issue is a complete lack of serial communication for provisioning, confirmed by timeouts from rnodeconf and the web flasher.

**To Reproduce**

-     Use a LilyGO T-Echo device.

-     Flash either an official RNode firmware (e.g., v1.83) or the Community Edition firmware (v1.75). The flashing process itself reports success.

-     Attempt to connect to the device using rnodeconf --info <your_serial_port> or the "Provision EEPROM" function in the web flasher.

**Expected Behavior**

The T-Echo should be recognized and configurable by rnodeconf and the web flasher immediately after the firmware is installed.

**Logs & Screenshots**

The command `rnodeconf --info <your_serial_port>` consistently fails with the same output for both firmware editions after a timeout:

```
[17:00:24] Opening serial port /dev/cu.usbmodem101...
[17:00:27] Serial port opened, but RNode did not respond. Is a valid firmware installed?
Got invalid response while detecting device
```

**System Information**

  OS and version: macOS

   Python version: Python 3.x

   Program version:

-         RNode Firmware (Official): All tested versions from v1.81 to latest.

-         RNode Firmware (Community Edition): v1.75.

-         rnodeconf: Latest version installed via pip.

**Additional context**

This issue occurs with both the official and Community Edition firmware lines. Standard troubleshooting steps (waiting after flash, manual resets, different USB cables/ports) have been performed without success.

The core of the bug is the failure to establish a command/response serial connection after the flash.

ronangaillard, 2025-08-24:
I noticed an early log that was hard to catch as it was send to serial a few ms after serial becomes available.

This is the error :

```
assertion "block < lfs->cfg->block_count" failed: file "/Users/ronan/Library/Arduino15/packages/adafruit/hardware/nrf52/1.7.0/libraries/Adafruit_LittleFS/src/littlefs/lfs.c", line 18, functio
```

I'm investigating

jacobeva, 2025-08-25:
Thanks for coming back and sharing your fix. I've run into this problem myself before. I'd happily accept a PR for a couple lines in the FAQ to explain this issue and how to fix it (please do not include the sketch in the PR, a link to this issue is okay though).

Closing as it's not a bug in the firmware.

closed 2025-08-25 by jacobeva